esphome/feature-requests #847: ledc & power_supply on the same pin

netztrip, 2020-08-11:
**Describe the problem you have/What new integration you would like**

I'd like to use the output component with _min_power_ & a _power_supply_ on the same pin.

**Please describe your use case for this integration and alternatives you've tried:**

Many PWM controlled devices like industrial fans or led driver use a 10-100% duty cycle logic where anything lower than 10% means off. In theory these devices could be controlled & turned off with only one GPIO. It seems this is currently not possible with esphome without an additional GPIO for a relay or some enable pin logic in hardware.

**Additional context**

One could also think of this as a bug - _set_level(0)_ should imho overide _min_power_. But I'm not sure if this breaks some other use cases.

OttoWinter, 2020-08-11:
`power_supply` is just there to drive another pin if an output is turned on (mostly for ATX power supplies). And when it activates it pulls the configued pin high, so using it for this use-case would be wrong.

`min_power` was iirc added for things like servos where the peripheral always expects some signal to check if there's anything communicating.

You could use a custom/template output platform to re-write the 0-100% to your own range, or (in this case probably the easiest) just don't min_power/power_supply at all. I mean 10% in a visual UI is quite little anyway and having that extra won't be a big issue.

WeekendWarrior1, 2021-08-14:
Can be closed as of:
https://github.com/esphome/esphome/pull/2013
https://github.com/esphome/esphome/pull/2107

closed 2022-07-06 by nagyrobi