esphome/feature-requests #848: Support tag scanning by introducing `homeassistant.tag`

balloob, 2020-08-11:
**Describe the problem you have/What new integration you would like**

Home Assistant 0.115 is adding a new `tag` integration. The goal is to make it easier for users to use tags to automate things. 

Tags are passed to the tag integration by calling the [`async_scan_tag`](https://github.com/home-assistant/core/blob/3fc2bae49ef07cea270ffddbc24e7a758aade806/homeassistant/components/tag/__init__.py#L109) method on the tag integration (in Python). From here on the user will be guided via a yet-to-be-made UI to automate it. 

```python
await tag.async_scan_tag(
    hass,
    tag_id,     # tag that was scanned
    device_id,  # of ESPHome device that scanned tag
)
```

We will need ESPHome integration to be able to call this method with data send in from an ESPHome device.

To do so, I would like to propose a new script syntax to ESPHome `homeassistant.tag`. When invoked, it will scan a tag inside Home Assistant.

Example how it can be used with a pn532:

```yaml
pn532:
  cs_pin: D0
  update_interval: 1s
  on_tag:
    then:
    - homeassistant.tag:
        tag: !lambda 'return x;'
```

**Please describe your use case for this integration and alternatives you've tried:**

Currently it can be implemented using events, and then having a custom integration set up on the backend that converts events to calling the tag integration.

```yaml
pn532:
  cs_pin: D0
  update_interval: 1s
  on_tag:
    then:
    - homeassistant.event:
        event: esphome.rfid_read
        data_template: 
          message: !lambda 'return x;'

```

**Additional context**

Gio-dot, 2020-08-11:
This is great! I would like to suggest to mostly use Wiegand protocol instead of various pn532-rc522 etc in ESPhome side. This because Wiegand is a specific protocol for this purpose (access control etc.) and there are a lot of cheap but very good readers on the market specifically made to be used even on the outside. I asked some months ago to integrate this protocol in Esphome, and others before..
https://github.com/esphome/feature-requests/issues/703
https://github.com/esphome/feature-requests/issues/211

I'm not a software specialist, so i'm not able to create custom component for Esphome. I actually use a Wiegand reader connected to an Arduino that read tags (using Monkeyboard library https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino), then Arduino send read code to an esphome device (via serial) that send it to Home assistant using **text sensor**. In Home assistant i use a binary sensor template to compare read code.

[image]

adonno, 2020-08-11:
I think we shouldn't limit to pn532 there are several devices that can currently read tags nfc/rfid and that are supported into esphome

@Gio-dot  mind hitting me up on discord? I have several questions on that device  :adonno#8391

glmnet, 2020-08-11:
> we shouldn't limit to pn532

Whats supposed to be the tag data? The way @balloob request the feature to be implemented it really doesn't matter if it comes from an NFC reader or any other ESPHome automation, so all current and future NFC readers will be supported this way.

adonno, 2020-08-11:
> > we shouldn't limit to pn532
> 
> Whats supposed to be the tag data? The way @balloob request the feature to be implemented it really doesn't matter if it comes from an NFC reader or any other ESPHome automation, so all current and future NFC readers will be supported this way.

yeah right my bad I misread it

balloob, 2020-08-11:
As long as it's a string of data it will be fine for the new tag integration in Home Assistant.

closed 2020-09-16 by OttoWinter

nardev, 2020-09-21:
Interesting, i didn't know for all this discussions before, i should try ESPHome for ESP-RFID boards.

Anyone interested, go check that out. https://github.com/esprfid/esp-rfid