esphome/feature-requests #849: TCS34725 RGB Color Sensor to RGB output to control RGB leds.

cromulus, 2020-08-11:
**Describe the problem you have/What new integration you would like**
I'd like to be able to use the output of this color sensor: TCS34725 RGB to change the color of an RGB Led strip. Currently, the output of percentages is not tremendously valuable. I know I can pre-process the output, however, it would be nice not to have to do that pre-processing.

**Please describe your use case for this integration and alternatives you've tried:**
I've tried to implement a filter that converts the output of the TCS34725 RGB Color Sensor to RGB, but it's been tough going to do so in yaml.

**Additional context**

I've always wanted lights that match the illuminance of my windows, especially during sunrise and sunset. 

Thanks!

andreashergert1984, 2020-12-10:
can you provide the calculation you want to do?

swifty99, 2022-02-11:
just as a sidenote: 
I tried to calibrate the color of LED lights with this sensor. The results were near bitperfect, however the visible colors ugly at best. Wavelength of sensors and LEDs probably do not match. So handle with care.