esphome/feature-requests #850: [Request] ESP8266Audio

keith721, 2020-08-11:
**Describe the problem you have/What new integration you would like**
platformIO ESP8266Audio component 
https://platformio.org/lib/show/1964/ESP8266Audio/

**Please describe your use case for this integration and alternatives you've tried:**
ESPHome-HomeAssistant-integration for audio notifier devices.  I currently use Rhasspy voice synthesis for these, but want to use a better-integrated and lighter-weight solution that is a smaller device.  Alexa, Google, and other online services are not acceptable. This should be entirely local to my home LAN.

**Additional context**
Hardware is ESP8266 (ESP12f / Wemos D1 mini) + LM386 audio amplifier driving a small 3-inch speaker.  Home Assistant automation will submit an MP3 or WAV file name to the ESP8266 to download and play the audio file.

timotoots, 2020-08-11:
Or integrate Snapcast so you could just push any audio to it.
https://mjaggard.github.io/snapcast/

keith721, 2020-08-31:
Doh, I'm a dummy.  :(  

I ordered DFPlayer modules from DFRobot, and will simply play specific MP3 files, instead.  Sorry for the bother.

closed 2020-08-31 by keith721