esphome/feature-requests #851: Support for Adafruit_AMG8833 8X8 pixel Thermal Camera

IsaTTeN, 2020-08-12:
**Describe the problem you have/What new integration you would like**
Adafruit AMG8833 Support. The AMG8833 is a Thermal Camera with a resolution of 8x8 pixel.
**Please describe your use case for this integration and alternatives you've tried:**
Since it measures temperature, It can detect a human from a distance of up to 7 meters (23 feet)

**Additional context**

I tried to do a custom component but I get errors. This is what I have done (this is my first custom component, don´t make fun of me)

`#include "esphome.h"
#include "Adafruit_AMG88xx.h"

class AMG88xxCustomSensor : public PollingComponent, public Sensor {
 public:
  Adafruit_AMG88xx amg;

  AMG88xxCustomSensor() : PollingComponent(15000) {}
  
  void setup() override {
    Wire.begin();
    amg.begin();
  }

  void update() override {
    float pixels[AMG88xx_PIXEL_ARRAY_SIZE];
    amg.readPixels(pixels);
    publish_state(pixels);
  }
};`

`esphome:
  name: test
  platform: ESP8266
  board: nodemcuv2
  includes:
    - AMG88xx_custom_sensor.h
  libraries:
    - "Adafruit AMG88xx Library"
    - "SPI"

wifi:
  ssid: !secret ssid
  password: !secret wifipass

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret captive_ssid
    password: !secret captive_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret apipass

ota:
  password: !secret apipass

i2c: 
  sda: D2
  scl: D1
  scan: True
  
sensor:
   - platform: custom
    lambda: |-
      auto my_sensor = new AMG88xxCustomSensor();
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      name: "Sensor AMG8833"`

[https://github.com/adafruit/Adafruit_AMG88xx](url)
[https://www.adafruit.com/product/3538](url)

DylanDonkersgoed, 2020-08-14:
I've actually been trying to do exactly what you are since last night, though I'm using a knockoff sensor. I'm guessing that like me you're trying to use the "Adafruit AMG88xx Library". I was able to resolve the SPI issue by including it under libraries, e.g.:

```
esphome:
  name: heatcam1
  platform: ESP32
  board: lolin32
  includes:
    - custom_amg8833.h
  libraries:
    - "Adafruit AMG88xx Library"
    - "SPI"
```

Perhaps it will work better for you.

Unfortunately I then ran into other issues that seem to be related to SD cards somehow. I think it's something to do with a dependency of some common library that Adafruit brings in. I elected to use the SparkFun library instead. It's not as popular but had the significant plus of actually compiling for my board. If the Adafruit library doesn't work for you either feel free to message me and I can send you what I've got so far, though it isn't really much.

IsaTTeN, 2020-08-14:
Hello @DylanDonkersgoed !

Yes! I had the same SD issue! So, were you able to integrate into esphome?. Could you share your progress?

Thanks

DylanDonkersgoed, 2020-08-14:
@IsaTTeN  It's not really integrated in any meaningful way yet but I *do* have it reading values from the sensor which I guess is the main thing. Here's what I've got, I'm sure you could adapt it to your use case:

```cplusplus
#include "esphome.h"
#include "SparkFun_GridEYE_Arduino_Library.h"

class AMG8833 : public PollingComponent, public Sensor {
 public:
  AMG8833() : PollingComponent(1000) {}

  float pixels[64];
  GridEYE grideye;

  void setup() override {
    ESP_LOGD("custom", "Starting up AMG8833 sensor");
    grideye.begin();
    delay(1000);
  }

  void update() override {
    for(unsigned char i = 0; i < 64; i++){
      pixels[i] = grideye.getPixelTemperature(i);      
    }

    for (unsigned char i = 0; i < 64; i++) {
      char buffer[8];
      ESP_LOGD("custom", dtostrf(pixels[i], 3, 2, buffer));
    }
  }

};
```

and of course you need the appropriate includes, i2c settings and sensor settings in your yaml file:
```yaml
esphome:
  name: heatcam1
  platform: ESP32
  board: lolin32
  includes:
    - custom_amg8833.h
  libraries:
    - "SparkFun GridEYE AMG88 Library"

i2c:
  # Change as per your preference
  sda: 19
  scl: 22
  scan: True

sensor:
  - platform: custom
    lambda: |-
      auto amg8833 = new AMG8833();
      App.register_component(amg8833);
      return {amg8833};

    sensors:
      name: "AMG8833"
```

IsaTTeN, 2020-08-15:
I got some errors in compling time, just in the line : 
`ESP_LOGD("custom", dtostrf(pixels[i], 3, 2, buffer)); `

I changed the code into this and now I can see the values of the camera in the log, but when I try to publish it, I get errors again in compilation
`#include "esphome.h"
#include "SparkFun_GridEYE_Arduino_Library.h"

class AMG8833 : public PollingComponent, public Sensor {
 public:
  AMG8833() : PollingComponent(2000) {}
  pixels = {};
  GridEYE grideye;

  void setup() override {
    ESP_LOGD("custom", "Starting up AMG8833 sensor");
    grideye.begin();
    delay(1000);
  }

  void update() override {
    pixels.clear();
    for(unsigned char i = 0; i < 64; i++){
      pixels.push_back(grideye.getPixelTemperature(i));
      ESP_LOGD("custom", "The value of sensor is: %f", this->grideye.getPixelTemperature(i));     
    }
    publish_state(pixels);    
  }
};`

brotherdust, 2020-10-10:
Has anyone on this thread had any success? I ended up having to use Micropython and MQTT to integrate with HA. I'd prefer to use ESPHome because that's what everything else I have uses.

wettermann32, 2020-11-06:
Hi all,

find attached some modifications to your code to publish values. I do not know how to publish arrays, anyhow you could use mqtt since you can access the global variable pixels[] in a lambda function in your esphome.yaml

`#include "esphome.h"
#include "SparkFun_GridEYE_Arduino_Library.h"

static float maxtemp;
static float mintemp;
static float avgtemp;
static float devtemp;
static float pixels[64];

class AMG8833 : public PollingComponent, public Sensor {
 public:
  AMG8833() : PollingComponent(1000) {}
  
  Sensor *max_temp = new Sensor();
  Sensor *min_temp = new Sensor();
  Sensor *avg_temp = new Sensor();
  Sensor *dev_temp = new Sensor();
  GridEYE grideye;

  void setup() override {
    ESP_LOGD("custom", "Starting up AMG8833 sensor");
    grideye.begin();
    delay(1000);
  }

  void update() override {
	maxtemp = -500.0;
	mintemp = 500.0;
	avgtemp = 0.0;
    for(unsigned char i = 0; i < 64; i++){
      pixels[i] = grideye.getPixelTemperature(i); 
      if(pixels[i] > maxtemp) maxtemp = pixels[i];
      if(pixels[i] < mintemp) mintemp = pixels[i];
	  avgtemp = avgtemp + pixels[i];
    }
    avgtemp = avgtemp/64;
	devtemp = grideye.getDeviceTemperature();
    //for (unsigned char i = 0; i < 64; i++) {
    //  char buffer[8];
    //  ESP_LOGD("custom", dtostrf(pixels[i], 3, 2, buffer));
    //}
	max_temp->publish_state(maxtemp);
	min_temp->publish_state(mintemp);
	avg_temp->publish_state(avgtemp);
	dev_temp->publish_state(devtemp);
	
	ESP_LOGD("amg8833 Thermal Sensor","Published new values to frontend");
	ESP_LOGD("amg8833", "max_temp %.1f min_temp %.1f avg_temp %.1f dev_temp %.1f", maxtemp,mintemp, avgtemp, devtemp);
  }

};`
find here the yaml for it:
`      
sensor:
  - platform: custom
    lambda: |-
      auto amg8833 = new AMG8833();
      App.register_component(amg8833);
      return {amg8833->max_temp, amg8833->min_temp, amg8833->avg_temp, amg8833->dev_temp};

    sensors:
      - name: "Heatcam_max"
        unit_of_measurement: °C
        accuracy_decimals: 1
        id: max_temp
      - name: "Heatcam_min"
        unit_of_measurement: °C
        accuracy_decimals: 1
        id: min_temp
      - name: "Heatcam_avg"
        unit_of_measurement: °C
        accuracy_decimals: 1
        id: avg_temp
      - name: "Heatcam_dev"
        unit_of_measurement: °C
        accuracy_decimals: 1
        id: dev_temp
`

aerodolphin, 2021-07-20:
Any news about this sensor?

closed 2022-07-01 by nagyrobi

definitio, 2022-07-01:
Why has it been closed? Custom component isn't good alternative to native integration.

nagyrobi, 2022-07-01:
Ask the author of custom component to submit it as a PR to ESPHome repo.