esphome/feature-requests #852: Update All function in background or via command line

poldim, 2020-08-13:
**Describe the problem you have/What new integration you would like**
Running update all from the dashboard requires dashboard tab to stay open and is pretty resource intensive.  

**Please describe your use case for this integration and alternatives you've tried:**
Adding an update all in the background function would eliminate the need for both of these.  Alternatively, it could be handled by executing a command line that would perform similar functionality.

rradar, 2020-08-13:
>  Alternatively, it could be handled by executing a command line that would perform similar functionality.

this exists already `esphome ... update-all`

poldim, 2020-08-14:
> > Alternatively, it could be handled by executing a command line that would perform similar functionality.
> 
> this exists already `esphome ... update-all`

I checked the docs but didn't find anything: https://esphome.io/guides/cli.html

I checked the help in the CLI and it is listed, but when running it, I got an error.  Might be user error though....

```
esphome /config/esphome_bbq_bot_v1.yaml update-all
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/usr/src/app/esphome/__main__.py", line 554, in main
    return run_esphome(sys.argv)
  File "/usr/src/app/esphome/__main__.py", line 523, in run_esphome
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/usr/src/app/esphome/__main__.py", line 362, in command_update_all
    files = list_yaml_files(args.configuration[0])
  File "/usr/src/app/esphome/util.py", line 250, in list_yaml_files
    files = filter_yaml_files([os.path.join(folder, p) for p in os.listdir(folder)])
NotADirectoryError: [Errno 20] Not a directory: '/config/esphome_bbq_bot_v1.yaml'
```