esphome/feature-requests #853: Deep sleep with MOSFET as circuit switch control

ruimarinho, 2020-08-13:
**Describe the problem you have/What new integration you would like**

I'm looking to control an ESP32 with deep sleep and an ultrasonic HR-S04 sensor via a MOSFET to cut power to the sensor whenever the ESP32 enters deep sleep. The circuit will be battery powered and I want to extend its longevity as long as possible. The quiescent current of the HR-S04 sensor is less than 2mA, but that's pretty significant for a battery-powered circuit.

A similar approach is documented at https://diyi0t.com/soil-moisture-sensor-tutorial-for-arduino-and-esp8266/:

[image]

Basically, I believe I only need to tell esphome to hold the gate of the MOSFET high as soon as possible after coming back from deep sleep to keep the sensor power on, and then low just before re-entering deep sleep.

On https://forum.arduino.cc/index.php?topic=216079.0, it is also suggested to disconnect the trigger pin otherwise the sensor is still able to draw power.

Is this something possible to achieve with esphome? I was thinking using `esphome.on_boot` hook and then call `switch.turn_on: mosfet`, where `mosfet` is defined as:

```yaml
switch:
  - platform: gpio
    pin: GPIO4
    id: mosfet
```

Even if this works, how would I switch the mosfet off before the ESP32 enters deep sleep?

Thanks!

ssieb, 2020-08-14:
Don't use the automatic deep sleep.  Use a timer trigger of some sort (various options) and turn off the pin before manually triggering deep sleep.  Something to check though is if you even need to do this.  Will the pin be still held high during deep sleep?  Also, questions like this would be better asked in the Discord channel.  A lot more people available to answer questions there.  I only saw this because I happened to be looking for something else.

nagyrobi, 2022-07-07:
Please head on to Discord for quick questions like this.

closed 2022-07-07 by nagyrobi