esphome/feature-requests #857: TM1637 additional controls

SnowyTerror, 2020-08-15:
**Describe the problem you have/What new integration you would like**
The ability to choose the brightness of the TM1637 display as there currently isn't one available.

**Please describe your use case for this integration and alternatives you've tried:**
This would allow further control to for example reduce the brightness to it's lowest being "1" when in low-light situations where it's not required to be as bright or if it doesn't need to be a main focus as there is no one in the room.

**Additional context**
No further context required.

glmnet, 2020-08-16:
From docs: 

> intensity (Optional, integer): The intensity with which the TM1637 should drive the outputs. Range is from 0 (least intense) to 7 (the default).

randybb, 2020-08-16:
Just tested and it works. But good idea, probably i will use it for my clock as well.

SnowyTerror, 2020-08-16:
Hi @glmnet, you would be right in this regard but this can only be set as a static variable, I don't see any option to be able to have this changed from HA at will

glmnet, 2020-08-16:
usually we update the brightness on the display lambda, is convenient and "easier" to implement, however it's usually not well documented...

try doing

````cpp
it.set_intensity(3);
````

in your lambda, of course you can give a variable there

SnowyTerror, 2020-08-16:
Again this lets you set an initial variable and doesn't allow you to change from HA after the fact of uploading the initial code.

To clarify I would like to be able to control the intensity of the TM1637 from HA as for example an entity or to include in an automation.

randybb, 2020-08-16:
It is doing exactly what it should do, in my test case switching to low brightnes while showing outside temperature:
`yaml
display:
  - platform: tm1637
    id: tm1637_display
    clk_pin: GPIO5
    dio_pin: GPIO4
    update_interval: 500ms
    lambda: |-
      static int i = 0;
      i++;
      if (id(alarm)) {
        if ((i % 2) == 0)
          it.print("8 8 ");
        else
          it.print(" 8 8");      
      } else {
        if (((i / 10) % 2) == 0) {
          it.set_intensity(7);
          if ((i % 2) == 0)
            it.strftime("%H.%M", id(ha_time).now());
          else
            it.strftime("%H%M", id(ha_time).now());
        } else {
          it.set_intensity(1);
          it.printf("%.0f~C", id(temperature).state);
        }
      }
`

SnowyTerror, 2020-08-18:
Hi Randy, I would be referring to where the variable is set in ESPHome sort of how yours is but then within Home Assistant I can at anypoint pick it up as an entity to for example be on the overview page so I can adjust it from 1 - 7 for example in a dropdown list or change it based on other information that is put into Home Assistant.

glmnet, 2020-08-18:
Yes. What you want to do is perfectly doable but not a feature request imo as this requieres some "coding"

As it requieres "coding" there might be several ways of doing it. 

I myself would create a global variable in esphome to store this brightness 

Also you'll need a service call from home assistant to update this variable. I haven't don't much this but I believe the helpers in home assistant will be needed 

I don't know if an alternative to the global/ service call could be a sensor, platform home-assistant. May be that works too. 

If you want quicker help the discord channel is very active on this kind of questions.

SnowyTerror, 2020-08-18:
Thank you, I will close this request.

closed 2020-08-18 by SnowyTerror