esphome/feature-requests #859: IR Remote Climate - Daikin, add missing modes

HA-TB303, 2020-08-16:
**Describe the problem you have/What new integration you would like**

The new IR Remote Climate daikin component (BETA) works great, but is missing an ECO, POWERFULL and QUIET mode.

**Additional context**
Here is a picture of my remote:
[image]

If I can do ANYTHING to help make this happen, please give me a heads-up!

jnimmo, 2022-05-27:
Also would be great to support different temperature steps (0.5 degrees instead of whole degrees).

cuongpl, 2024-07-16:
Waiting for this update to be released!

Shambler2, 2024-07-30:
+1 for this. I use Eco & Quiet together 100% of the time

closed 2026-10-11 by bdraco