esphome/feature-requests #861: Snapcast support

MayeulC, 2020-08-18:
**Describe the problem you have/What new integration you would like**
Integrating the following work with esphome: https://github.com/jorgenkraghjakobsen/snapclient

**Please describe your use case for this integration and alternatives you've tried:**
Create a free software, cheap sonos clone for multi-room audio.

**Additional context**
I would like to have a system that is as energy-efficient as possible, I think the esp32 might be suitable for this. Ideally, the workflow would be the following:

Materials needed:
1. ESP32
2. MA12070P or similar
3. Analog speakers
4. (battery if you want) power supply

Software needed: Snapcast server

Ideally, I would like the installation process to be as simple as soldering together the modules, specify it in the config, and flash, then have it pop in homeassistant.

A possible alternative would be to use separate bluetooth speakers. I am not sure how feasible this would be?

velaar, 2022-10-26:
It would be great to see this integrated. Sadly I lack the required knowledge to do so. I've recently started experimenting with LyraT 4.3 boards and this fork seems to be working well: [https://github.com/CarlosDerSeher/snapclient] - it looks more advanced than the original in the post above

Bram-diederik, 2023-05-10:
Hi, Just passing by. adding snapcast support will make music adaptive lights with home assistant possible.

h3ndrik, 2024-02-16:
Just to add some more recent forks of the snapclient:

* https://github.com/CarlosDerSeher/snapclient
* https://github.com/pschatzmann/arduino-snapclient

mads03dk, 2024-06-27:
I have begun working on something that has a slight possibility of turning into a Snapcast component. This is my first time ever developing components for esphome. Although I have experience with both Arduino and ESP-IDF I am by no means an expert in c++ which heavily limits this specific type of development with protocols, buffers, memory allocations and all the other low level stuff where I'm not that strong.

Dev fork: https://github.com/mads03dk/esphome

At the moment, I don't have the time required to complete this. This is heavily reflected in the catastrophe of a commit I just made to repo.

If anyone wants to help or take inspiration from my crappy sketch, feel more than free to do so :)

What works:
- Esphome .yaml config
- Connection to server
- Base message serialization/deserialization
- Hello message to the server
- Codec message handling
- Time message receiving (correction doesn't work yet)

yggi, 2024-09-29:
I would love to have an esphome snapcast component. Synced multi-room audio right out of the sensor nodes! Is there any progress / timeline on this? I might be able to put some work in myself, although I am in a similar situation as @mads03dk  and would be very grateful for some handholding.

saeugetier, 2024-10-14:
@mads03dk awesome work! are you planning to work on synchronization of streams?

mads03dk, 2024-10-15:
@saeugetier 
If I had the time and skill to do it, then yes. But unfortunately, I'm currently depleted of criteria no. 1 and lack criteria no. 2.
Perhaps sometime during the winter, if I'm not too booked out and have the energy to do it. But I'd be more than happy if someone else would take a look at the integration in the meantime

Chreece, 2024-11-23:
There is an [arduino](https://github.com/pschatzmann/arduino-snapclient) port for snapclient, would it make sense to use it as library for esphome? Maybe this would not need much work to port here...

Californian, 2025-02-02:
This would be huge to allow the Home Assistant Voice PE hardware to be used as the basis for a real multiroom audio system!

nagyrobi, 2025-02-24:
And these seem to be made for ESP-IDF, not arduino

velaar, 2025-02-24:
Snapcast support will be great as I assme at present there is no path for HomeAssistant VE to have Multi-room audio.

yggi, 2025-04-03:
@luar123 thx for integrating this, really looking forward to have snapcast natively in esphome! ❤

Do you need any help with testing, documentation, etc?

steel4me, 2025-10-01:
AFA Resonate is better and working correctly; I go for it. Just need something that is working correctly.

Hedda, 2025-10-08:
FYI, FutureProofHomes developer for their ESPHome fork ([gnumpi](https://github.com/gnumpi)) has added their own Snapcast client implementation to that repo during the last few months, see:

- https://github.com/FutureProofHomes/Satellite1-ESPHome/pulls?q=is%3Apr+snapcast

and 
-  https://github.com/FutureProofHomes/Documentation/pulls?q=is%3Apr+snapcast