esphome/feature-requests #867: Organise devices in Folders

PierreScerri, 2020-08-24:
**Describe the problem you have/What new integration you would like**

It would be nice and more efficient to organise my ESPHome devices into folders.

**Please describe your use case for this integration and alternatives you've tried:**

As the number of ESPHome devices in my home increases, it is becoming more tedious to find a specific device.

One could also possibly use the 'Update' option on a folder instead of a 'One or All' update. 

'Update all' is not really useful if you have updated an !include file for a bunch sonoff duals and you only need to update those and not all the other Basics, Minis and Wemoses.

Having a grid display instead of a long list would also help.

Thank you

**Additional context**

--

thedexboy, 2020-08-24:
I've installed the dev version yesterday and it has a grid layout. Way better than before.

PierreScerri, 2020-08-24:
My ESPHome is an addon (ver 1.14.5) to Hassio. I don't think that it gives me the option to install the dev version.

glmnet, 2020-08-25:
Yes, beta and dev, both have the new feature and both are available as add-ons, you can install them, check docs page https://esphome.io/guides/faq.html#how-do-i-update-to-the-latest-beta-release

PierreScerri, 2020-08-26:
That is already a lot better.  Thank you.

glmnet, 2020-12-28:
If somebody is going to work on this, I believe the folder should not be virtual but follow the filesystem folder structure, this will make more sense as you can use packages / fonts / secrets, etc. and include files in other folders.

jbuszkie, 2020-12-29:
I hope someone can work on it... 🤞

lcavalli, 2021-11-22:
> If somebody is going to work on this, I believe the folder should not be virtual but follow the filesystem folder structure, this will make more sense as you can use packages / fonts / secrets, etc. and include files in other folders.

I agree. Having support for sub-folders at filesystem level it will be possible to archive additional files alongside the yaml configuration, like PCB gerber files, pdf with protocol specification, .gitignore files, additional .h and .cpp source code without messing up with other nodes.

maslyankov, 2023-12-19:
Any news on this topic?

sjpilot, 2024-01-04:
A partial workaround is to use the !include yaml per https://esphome.io/guides/configuration-types.html#yaml-insertion-operator. Unfortunately you will still need a file per device but it can be a placeholder to include a file in a git repository folder.
`<<: !include testproject/testdevice.yaml
`

w531t4, 2026-09-27:
I have 32 esphome devices. Having 32 files in /config/esphome makes it difficult to find the right one, and requires excess scrolling. +1