esphome/feature-requests #868: Climate add vertical (and horizontal) flaps position.

F43RY, 2020-08-26:
**Describe the problem you have/What new integration you would like**

Hi, I'd like to have the "flaps" position function in climate component. Now are only available VERTICAL and OFF.

**Please describe your use case for this integration and alternatives you've tried:**

I configured [geoffdavis](https://github.com/geoffdavis/esphome-mitsubishiheatpump) esphome custom Climate code in my ESP01 to get my mitsubishi heat pump work. This code includes [SwiCago](https://github.com/SwiCago/HeatPump/tree/master/src) code that implements every function of my heat pump. Among the heat pump functions is there the vertical position of flaps that can operate as swing but also can be set at pre-estabilished positions (1,2,3,4,5, SWING, AUTO). With actual configuration I can only set VERTICAL (SWING) or OFF (AUTO) ignoring the remaining positions.
Is it possible to add a new ENUM in climate_mode.h (also related functions in traits.h and climate.h) containing the 7 values (1,2,3,4,5,SWING, AUTO)  for vertical flap positions?
A further request could be add the same ENUM for Horizontal flap positions since a few models have also this function.

**Additional context**

Thx F43RY

maretodoric, 2021-06-16:
Uuum. I'm also interested about this. Any ideas for a workarounds at least?

F43RY, 2021-06-26:
The following PR could be a good start point
[PR1968](
https://github.com/esphome/esphome/pull/1968)

Pablo-Ortiz-Lopez, 2023-07-13:
In my PR #5097 , I have implemented the vane settings mapped as the supported 'swing' values:

- SWING_HORIZONTAL -> Vane high
- SWING_VERTICAL -> Vane low
- SWING_BOTH -> Vane moving
- SWING_OFF -> Vane in the middle

duhow, 2025-03-30:
With recent addition https://developers.home-assistant.io/blog/2024/12/03/climate-horizontal-swing/ ,
now we should request to implement SWING_HORIZONTAL_MODE into ESPHome.