esphome/feature-requests #871: Add support for ESP32-S2-Kaluga-1

MadMacMad, 2020-08-28:
**Describe the problem you have/What new integration you would like**

would love to add the Kaluga with LCD & Touch + Camera and Speaker to my HASS.io 
think most is there but LCD, Touch, Camera and Speaker need drivers...

**Please describe your use case for this integration and alternatives you've tried:**

can use the Kaluga with ADA IDE and it works too but integration with ESPHome would best case

**Additional context**

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp32-s2-kaluga-1-kit.html
The ESP32-S2-Kaluga-1 kit consists of the following boards:

Main board: ESP32-S2-Kaluga-1
Extension boards:

ESP-LyraT-8311A audio player
ESP-LyraP-TouchA touch panel
ESP-LyraP-LCD32 3.2” LCD screen
ESP-LyraP-CAM camera board

we need support for this amazing device for this price ... plz 
thank you!

randybb, 2020-08-28:
ESP32-S2 is still not supported by platformio https://github.com/platformio/platform-espressif32/issues/349

And for the other peripherals:
- ESP-LyraT-8311A audio player - I2S is not supported https://github.com/esphome/feature-requests/issues/599
- ESP-LyraP-TouchA touch panel - is supported https://next.esphome.io/components/binary_sensor/esp32_touch.html
- ESP-LyraP-LCD32 3.2” LCD screen - display uses probably ILI9341 still not officially supported https://github.com/esphome/feature-requests/issues/51 / touch driver NS2016 is not supported
- ESP-LyraP-CAM camera board - Both camera sensors are supported https://next.esphome.io/components/esp32_camera.html

askpatrickw, 2020-09-18:
Seems like this is in release 2.0.0 as a Beta.
https://github.com/platformio/platform-espressif32/releases/tag/v2.0.0

TechIsCool, 2020-10-01:
```ESP32-S2 Beta firmware is not compatible with the ESP32-S2 production SoC, and not all features are supported.``` Still looks like a little bit of time before its supported.

mannro, 2020-11-09:
still not possible to use esp32s2 ?

randybb, 2020-11-09:
@mannro Since the latest version of PlatformIO is still the same beta 2.0.0, that means no. As ESPHome depends on a stable release of PlatformIO, you need to ask them.

misery, 2020-12-26:
This platformio.ini works for me with an example project.
Of course it is experimental but I want to try that current dev state on my dev machine. How can I configure an esphome yaml to try that branch/versions? So I can try to fix some problems with it.

https://github.com/espressif/esp-idf/releases/tag/v4.2
https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2

```
[env:esp32s2]
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
platform = espressif32
board = nodemcu-32s
framework = arduino
board_build.mcu = esp32s2
```

misery, 2020-12-26:
Ok, tried it like this...

```
esphome:
  name: espdemo
  platform: ESP32
  board: nodemcu-32s
  #arduino_version: 1.0.4
  platformio_options:
    board_build.mcu: esp32s2
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
```

If I comment ```WiFi.onEvent(f);``` out it compiles and my ESP32-S2 started with my esphome project. But of course.... WiFi isn't working. 

Needs esphome a fix here or is it a misconfiguration?

Callback:
```
typedef std::function<void(arduino_event_id_t event, arduino_event_info_t info)> WiFiEventFuncCb;
```

```
void WiFiComponent::wifi_event_callback_(system_event_id_t event, system_event_info_t info)
```

```
src/esphome/components/wifi/wifi_component_esp32.cpp: In member function 'void esphome::wifi::WiFiComponent::wifi_pre_setup_()':
src/esphome/components/wifi/wifi_component_esp32.cpp:467:17: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind<void (esphome::wifi::WiFiComponent::*(esphome::wifi::WiFiComponent*, std::_Placeholder<1>, std::_Placeholder<2>))(system_event_id_t, system_event_info_t)>&)'
   WiFi.onEvent(f);
                 ^
In file included from /home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiSTA.h:28,
                 from /home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFi.h:32,
                 from src/esphome/components/wifi/wifi_component.h:13,
                 from src/esphome/components/wifi/wifi_component_esp32.cpp:1:
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:145:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:145:21: note:   no known conversion for argument 1 from 'std::_Bind<void (esphome::wifi::WiFiComponent::*(esphome::wifi::WiFiComponent*, std::_Placeholder<1>, std::_Placeholder<2>))(system_event_id_t, system_event_info_t)>' to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'}
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:146:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:146:21: note:   no known conversion for argument 1 from 'std::_Bind<void (esphome::wifi::WiFiComponent::*(esphome::wifi::WiFiComponent*, std::_Placeholder<1>, std::_Placeholder<2>))(system_event_id_t, system_event_info_t)>' to 'WiFiEventFuncCb' {aka 'std::function<void(arduino_event_id_t, arduino_event_info_t)>'}
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:147:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/andre/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/libraries/WiFi/src/WiFiGeneric.h:147:21: note:   no known conversion for argument 1 from 'std::_Bind<void (esphome::wifi::WiFiComponent::*(esphome::wifi::WiFiComponent*, std::_Placeholder<1>, std::_Placeholder<2>))(system_event_id_t, system_event_info_t)>' to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'}
Compiling .pio/build/espdemo/src/esphome/core/esphal.cpp.o
*** [.pio/build/espdemo/src/esphome/components/wifi/wifi_component_esp32.cpp.o] Error 1
```

misery, 2020-12-26:
Ok, fixed the WiFi problem.... my ESP32-S2 works now: https://github.com/esphome/esphome/pull/1433

misery, 2021-02-05:
https://github.com/platformio/platform-espressif32/releases Version 3 has support for ESP32-S2 now.

eisenhowerj, 2021-09-22:
Its available in the platformio dependency currently shipping with latest esphome:

```﬌ source venv/bin/activate

﬌ pip install esphome
...snip...

﬌ platformio boards | grep -i ESP32S2
featheresp32-s2                  ESP32S2  240MHz       4MB      320KB   Adafruit ESP32-S2 Feather Development Board
esp32-s2-kaluga-1                ESP32S2  240MHz       4MB      320KB   Espressif ESP32-S2-Kaluga-1 Kit
esp32-s2-saola-1                 ESP32S2  240MHz       4MB      320KB   Espressif ESP32-S2-Saola-1

﬌ pip list | grep platf
platformio         5.2.0```

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996

closed 2022-07-01 by nagyrobi