esphome/feature-requests #872: Add support for Xiaomi BLE Temperature and Humidity sensor (LYWSDCGQ) on ESP32

vladacr, 2020-08-28:
**Describe the problem you have/What new integration you would like**

Please add support for the LYWSDCGQ [XIAOMI Mijia Bluetooth Smart Electric Digital Thermometer Hygrometer](https://www.banggood.com/3Pcs-XIAOMI-Mijia-Bluetooth-Smart-Electric-Digital-Thermometer-Hygrometer-2-p-1595118.html?rmmds=myorder&cur_warehouse=CN) 
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

wizu, 2020-08-31:
@vladacr it's already supported at dev branch.
https://next.esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc

rradar, 2020-09-01:
as mentioned support is already present in next/dev version

closed 2020-09-01 by rradar