esphome/feature-requests #874: Expose scripts to Homeassistant as script entity

hcoohb, 2020-08-29:
**Describe the problem you have/What new integration you would like**

It is possible to create scripts on ESPHome.
Homeassistant has support for a script class and offer to "execute" script entity by default:
[image]
From my search it seems that there is no way to fire a script on the ESPHome from HA.
Not sure if there is a technical reason not to, but that would prove super handy feature to have!
Maybe just setting the name to allow to have it exposed on the frontend?:
```yaml
script:
  - id: script_id
    name: "My script Name"
    then:
      ...
```

**Please describe your use case for this integration and alternatives you've tried:**

It would allow to fire a ESPHome script from the HA frontend, using the HA script `entity class` ("Execute" button by default).
An example I am trying to use it for it just transmitting one IR/RF signal through the remote transmitter. (It is not to control a switch or a light, but to program/initialise a RF plug) .

We can use as an alternative a switch and firing the script on `on` but using the entity script would make it clearer!

**Additional context**

glmnet, 2020-08-29:
I don't know enough to tell if what you want is possible. 

But I can tell a better way of doing it now is by using the api and make a service call from HA 

Check this https://esphome.io/components/api.html#user-defined-services

hcoohb, 2020-08-30:
Thanks for the quick answer @glmnet !

I agree, creating a service is another way of going around this.
However, because services do not appear on the device page of the integration in HA, unlike all other entities, I just think it would be much more clearer if we can expose scripts on the device page. (And it has the advantage that it can be added straight to lovelace with the push of a button)

glmnet, 2020-08-30:
Do you know if HA supports this? Is there an example?

E.g. I have a MikroTik router and I'm using a MikroTik integration, not the one included in HA but other integration. This integration exposes MikroTik scripts as Switches in HA. So this is why I say I don't know enough.

hcoohb, 2020-08-30:
Oh, I see what you mean... that it might not be possible in HA side to have script in devices...
You are right, so far I cannot see any devices having a script registered...
I will try to dig a bit more in the code and report what I find.

Thanks!

AJolly, 2024-05-24:
Any good way to do this these days?

ll3141, 2024-07-12:
you can use a template switch running the script from the on_turn_on action of the switch

it works, but it's pretty awful