esphome/feature-requests #875: extend i2c for 16 bit registers

torwag, 2020-08-30:
Currently, the implementation of the I2C interface works with 8bit address registers.
Some units like the VL53L1X use 16 bit address registers.
Thus, those devices cannot easily be integrated into esphome right now.

There are two major ways to address devices with more than 8 bit address registers.
First is the usage of n device addresses, which basically splits up the device in different register banks and hence allows to further utilize 8bit address registers. That can be seen quite often in I2C-based memory chips.
Second is the splitting of an 16 bit address into 2 x 8bit transmissions, order of MSB and LSB is device specific. After sending the 16 bit address the device awaits resp. sends the data.
To make it even more complicated some devices uses a combination of both methods. 

From there the question below arise:
1. How many sensors/devices are already known which can't be implemented right now due to the above issue.
2. If question 1 provides a large enough number, does it make sense to implement the 16 bit address registers in the i2c component (if numbers are very low, it might be better to do that in the component itself).
3. What would be the best way to implement it, taken that there are different possible ways to address 16 bit registers (see above). 
4. Overloading the member functions with a 16 bit version for a_register of the I2C classes would almost replicate the entire class, but it is still questionable how different ways of addressing could be exposed to higher level component implementations.   

**Additional context**

nagyrobi, 2022-07-08:
The mentioned PR ran on a timeout and was not merged. Not interested anymore?

CarsonF, 2023-03-28:
There's definitely still interest. I've worked on a people counting solution using this sensor, [Roode](https://github.com/Lyr3x/Roode). 
We are forced to use Arduino's `Wire` instead of esphome's i2c abstraction.
https://github.com/rneurink/VL53L1X_ULD/blob/master/src/platform/VL53L1X_i2ccoms.cpp

ssieb, 2023-03-28:
There's no reason to have to use `Wire`.  You just have to create the packet yourself and use the basic read and write methods.  That's all the register methods do.  They're just convenience methods to do the packing for you.

CarsonF, 2023-03-28:
😮 🤦‍♂️ That sounds so obvious now that you say that.

[This](https://github.com/esphome/esphome/pull/3052/files#diff-78e4c64d193eff14656a65821e0738f852535f5b8cb7b1c5b37e5c4b8464377dR83-R88) is on the right track to doing that right?

ssieb, 2023-03-28:
I guess so, but it seems overly complicated.  Are there likely to be registers over 16 bits?

ssieb, 2023-03-28:
I would just create new methods for the 16-bit registers.

CarsonF, 2023-03-28:
It was requested in discord that I don't do that. I don't remember by who...maybe Otto.

ssieb, 2023-03-28:
I don't see any mention there about creating new methods for the 16-bit registers.  You would need `read_register16`, `write_register16` and maybe add a new `I2CRegister16` class.  I don't see any need for more than that or allowing arbitrary sizes.

mrtoy-me, 2023-06-27:
> Got it thanks for the info. Perhaps I can PR this soon.

Hi Carson

If there is still interest, I have done a ESPHome component that uses ESPHome low level I2C read and write function to make some 16 bit addressing I2C functions like ssieb suggests. The component works fine and gives distance and range status. Let me know if you are interested in testing it.

RonaldTHerman, 2024-02-20:
The Adafruit rotary encoder 5880 requires a 16 bit address... High byte selects a module and the low byte the register in that module.

PIGPIO ERROR: INVALID I2C REGISTER [4400]; Valid range: 0-255   (0x00110030)

justinschuldt, 2024-07-04:
This was done via https://github.com/esphome/esphome/pull/4844