esphome/feature-requests #876: Add thermostat option for minimum circulation per time period

ptr727, 2020-09-01:
**Describe the problem you have/What new integration you would like**

The request is for the thermostat or bang_bang component.
The behavior is typical of fan settings that are used for air circulation and filtration even if there is no call for heating or cooling.

I want to control my home instant hot water circulation pump.
The pump circulates hot water in a loop so that there is always hot water when a faucet or shower is turned on.
The pump uses a fixed temperature aquastat (e.g. on at 85F off at 105F) that measures the return circulation pipe temperate, when the temperature drops the pump turns on, when water flows the instant hot water heater turns on, and heats the water in the circulation line, until a set temperature is reached, and the pump turns off.
I want to replace the hardwired aquastat with e.g. a ESP32 or TH16, and I want more control over the pump operation.

**Please describe your use case for this integration and alternatives you've tried:**

One problem is that the pump is in the garage, and in the summer the garage ambient temperature is higher than the aquastat minimum (e.g. 85F), and the circulation pump never turns on due to the return pipe being at ambient temperature. But, the water in the circulation pipe is cold, thus it takes a long time to get hot water.

I want to use the temperature sensor, but also specify the minimum runtime of the pump, such that no matter how warm the garage and plumbing in the garage gets, the pump will periodically turn on and circulate the water.

This mode is very similar to that of HVAC fans, that can be programmed to run a minimum number of minutes per time period to circulate or filter air. E.g. if the AC is on then the fan is running anyway, if the AC is off then the fan will still kick in to circulate air even if AC is not called.

Unlike the more common HVAC fan use case, I want to use the setting for the pump switch; turn the pump on at low temp, turn the pump off at high temp, make sure the pump runs x minutes out of y minutes.

**Additional context**

kbx81, 2020-09-13:
Reading through this, I understand what you are going for but I'm not sure this behavior makes sense to build into the thermostat component/integration; I think in general we try to minimize feature creep within components and, in many cases, extra functionality such as this can easily be accomplished with the use of an additional  component (that potentially already exists).

The way I would accomplish this myself would be to use `interval:` to toggle the switch that controls your pump on (and off?) at a frequency you set. For example, you could create a template switch that the `interval` action turns on; this template switch's `turn_on` action is to turn on the switch that controls your pump, `delay` for some duration, then turn off the switch that controls your pump. This alone gets you the "stirring" action you desire.

With this in place, you'd just need to work out the interaction between the `thermostat` component and the `interval` action(s) -- you don't want the `interval` to turn off the pump when the temperature has not yet reached its desired target. This could be accomplished with the use of a second template switch and a little ~~magic~~ boolean logic. 😄

ptr727, 2020-09-13:
Agree I could do it another way.

But, as I was thinking about the logic, it just started sounding exactly like my Ecobee thermostat fan air filtration function, where the purpose is a certain amount of minimum air circulation per time period.

So, air filtration could still be a good thermostat feature, or if you don't want the thermostat to have too many features, maybe a different type of component, but it should still be able to track fan runtime to get the desired circulation per period.

adamj12b, 2020-10-09:
I would like to second this request, but for the use in an HVAC application.

It is much more efficient to move the air around using the fan for a period of time every hour(s) rather than to bring on heating or cooling and the fan just to find out once the air is circulated, the space is actually quite comfortable. It also cuts down wear and tear on the compressors starting in air conditioning and the furnace burner starting in boilers.

A second request to this would be on and off delays for the fan.

In commercial  HVAC control, we can program minimum on/off time for the fan as well as on/off delay to start the fan after the plenum has come up to temperature and to keep the fan running to cool the plenum before shutting off the fan.

kbx81, 2020-10-12:
I'd be interested to hear the input of @ssieb, possibly @glmnet and/or @jesserockz on this one. My personal take on ESPHome components/integrations is that they should provide basic functionality to keep the compiled code's footprint small; they can easily be integrated together in a variety of ways to implement more complex and/or complete functionality such as what is being described here.

> it just started sounding exactly like my Ecobee thermostat
We are conflating a (finished/complete) _device_ with an ESPHome _component_. This functionality is absolutely possible to implement within this ESPHome component itself, but at the expense of a larger (binary) footprint when it is compiled; it will consume space that some folks may wish to use for other purposes. I've personally already run into memory limitations on the ESP32 modules I use in the thermostats I've built myself.

Speaking of them, if you want delays around turn-on/turn-off of various functions, I have already implemented this by using ESPHome scripts. See the YAML configurations I have shared here: https://github.com/kbx81/ClimateSprinklerController/blob/master/ESPHome/esp_thermostat_nook.yaml
> In commercial HVAC control, we can program minimum on/off time for the fan as well as on/off delay
My scripts do more or less exactly what's described here -- it's just done outside of the component itself. In my personal opinion, this is better, because it allows the designer/user to implement exactly what they need vs. us trying to implement it in a way that pleases everybody who wants functionality beyond what the component itself provides. I personally like having the delay implemented _outside_ of the component because it allows me to display (to the user) the current action _immediately_ even though the output itself has not changed state yet. This is a better user experience because they can see what will happen instead of wondering "why isn't it doing anything?" Integrating the delay into the component would almost certainly hide what's going on, leading to confusion.

Regarding periodic air/water circulation, one question I have is this: what does the UI (in HA) look like to control this? Let's say the seasons change and I want to adjust the frequency and/or duration of the fan's operation. As things are now, there is climate-specific UI in HA that enables control of the upper and lower set points, six operating modes, nine fan modes and four swing modes. As it relates to periodic operation of fans/pumps, there is no UI available to control this aspect of the device's behavior, which leaves you to re-flashing your ESPHome device should you want to change this interval. That's not an ideal user experience. You could use ESPhome's `api` integration to create your own way to do this, but now you're back to writing your own code, both in ESPHome and arguably now in HA, too. Perhaps this isn't necessary -- it's just a thought/outstanding question I had and I'd love to hear more thoughts on it.

I'll be perfectly clear: I'm not opposed to this additional functionality. It's a great, very practical, even necessary idea...and I've even already done it myself. I'm just not sold on that it belongs integrated into the `thermostat`/`bang-bang` components in light of all the other things we already have in ESPHome that can be used to do this.

That all said, what if we had an "Advanced Thermostat" that had all the bells and whistles? It might be nice to keep this one as a basic/"slim" component and then have a "it does everything" component, as well. Does that seem like a nice idea?

ssieb, 2020-10-12:
The benefit of adding the functionality to the existing thermostat component instead of creating a new one is that we're just adding a small amount of code instead of completely duplicating all of the code and then adding a bit more.  The drawback is that there would be no HA UI to control the added functionality.  However, that doesn't get solved by making a new component either.  The suggestion of adding some extra separate components to add the extra functionality isn't very effective because they would need to get information from the thermostat component that doesn't exist right now.  And there's still the problem of how to make a usable UI.
There's definitely a case for keeping components simple, but in order for that to be useful, they need to either be directly extensible or provide enough hooks to be able to add the extra functionality separately.

GigaTech, 2020-10-28:
Hi Guys, I would like to second second this...

I have a compressor (refrigerant) and fan running fine from a thermostat, the compressor can only run for 5mins with 10mins off, the evaporator (cold bit) will stay cold for some time and the fan should stay on to circulate over the evaporator, cycle the compressor on/off until temp reached and leave fan on for a further x mins

And exactly the same for a heater, just because you have hit temp + any hysteresis and turn it off doesnt mean the is is still not producing heat, so you need the fan to keep circulating to prevent 'hot spots'

Also switching some compressors and heaters on/off quickly or within a short time could cause damage to equipment or start to pop fuses

The min 'rest' time should also be observed between cycles. ie. cooling hit temp 8mins ago now too hot, call for cold, wait 2mins for the timeout to pass.

just my thinking, something like this maybe, in the .yaml
```
- platform: thermostat
  name: "xController"
  id: x_control
  sensor: x_temp
  default_target_temperature_low: 10.0 °C
  default_target_temperature_high: 20.0 °C
  hysteresis: 5.0

  cool_action:
    - switch.turn_on: x_cool
      minrun: 3min
      maxrun: 5min
      nextrun: 10min
	  
    - switch.turn_on: x_fan
      extrarun: 5min

  heat_action:
    - switch.turn_on: x_heat
      minrun: 1min
      maxrun: 5min
      nextrun: 10min
	  
    - switch.turn_on: x_fan
      extrarun: 5min

  idle_action:
    - switch.turn_off: x_cool
    - switch.turn_off: x_heat
    - switch.turn_off: x_fan

```

If any of these prams omitted then default to 0 and disable the feature

bob-u, 2020-11-20:
(I am not a user yet just considering ESPhome) I just opened feature request for similar feature: https://github.com/esphome/feature-requests/issues/1008 and found this thread.  I understand that intention of developers is to reuse components therefore I think on_multi_click https://esphome.io/components/binary_sensor/index.html#binary-sensor-on-multi-click may work:

```
binary_sensor:
  - platform: gpio
    # ...
    on_multi_click:
    - timing:
        - OFF for at least 360s
        - ON for at least 5s
      then:
        - logger.log: "Double-Clicked"
```

Above seem to ensure that device is off for at least 5 minutes before it can be turned on.  What I am missing, I believe, is ability for the timer to survive reset/reboot to ensure that it does not erase the delay.

nagyrobi, 2022-07-01:
Options `min_cooling_off_time`, `min_cooling_run_time`, `min_heating_off_time`, `min_heating_run_time` were added to [Thermostat Climate Controller](https://esphome.io/components/climate/thermostat.html)

closed 2022-07-01 by nagyrobi

ptr727, 2022-07-01:
Could the funcitonality please be added to Climate and not to Thermostat, that way bang_bang also gets to use them, i.e. I am using the bang_bang controller for hot water recirc, not thermostat?