esphome/feature-requests #877: Add support for ATC_MiThermometer

danielkucera, 2020-09-02:
**Describe the problem you have/What new integration you would like**

I'd like to see ATC_MiThermometer supported in esphome.

**Please describe your use case for this integration and alternatives you've tried:**

The original firmware is supported but it requires complicated bind key generation and only sends the data once in 10 minutes.
The opensource firmware can be programmed to any interval.

**Additional context**

This has already been discussed a bit in closed issue #552 . I'm opening this one to not spoil the original issue.
All protocol info can be found here: https://github.com/atc1441/ATC_MiThermometer

tubalainen, 2020-09-02:
This is GREAT! I would love not to go through the token/hash process to hijack the MI communication.

Kuglikrug, 2020-09-02:
This would be amazing. Flashing atc1441's firmware works like a charm and integration with esphome would really help!!

danielkucera, 2020-09-02:
BTW, this seems like a 2hour job, I  can try to implement it today.

Calcu2006, 2020-09-05:
It was possible to implement ?

ahpohl, 2020-09-05:
Initial support for the [ATC MiThermometer](https://github.com/atc1441/ATC_MiThermometer) custom firmware can be found here: [https://github.com/ahpohl/esphome/tree/atc_mithermometer](https://github.com/ahpohl/esphome/tree/atc_mithermometer). I have currently implemented temperature, humidity and battery level in %. The custom firmware also broadcasts battery level in mV, which is not collected yet.

Configuration is like this:
```
esphome:
  name: basement
  platform: ESP32
  board: nodemcu-32s 

wifi:
  ssid: "your ssid"
  password: "your wifi password"

api:

logger:
  #level: VERY_VERBOSE

ota:

esp32_ble_tracker:
  scan_parameters:
    window: 120ms

web_server:
  port: 80

mqtt:
  broker: 192.168.178.55

sensor:
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:4E:16:78"
    temperature:
      name: "ATC Temperature"
    humidity:
      name: "ATC Humidity"
    battery_level:
      name: "ATC Battery-Level"
```
Download and compile a new firmware:
```
git clone https://github.com/ahpohl/esphome.git
cd esphome
git checkout atc_mithermometer
python -m venv venv
source venv/bin/activate
python setup.py install
mkdir Testing; cd Testing
esphome atc_mithermeter.yaml run
```

ahpohl, 2020-09-05:
@danielkucera  As the method how to generate a new bindkey and upload to the device is now known, did you meanwhile update your python script?

Now we have some more possibilities to use the Xiaomi LYWSD03MMC device with ESPHome:

1. flash the custom firmware and use the (not yet officially released)  `atc_mithermometer` component without encryption
2. use the firmware updater to just generate and upload a new bindkey and use the `xiaomi_lywsd03mmc` component
3. in case the firmware updater doesn't work (it only seems to work in Chrome and not in Firefox), it would be nice to also have another method like your standalone python script to upload a new bindkey
4. get the ESP to provision the device on first discovery (this would be possible now, but needs new code to actively login to the device and upload a new key)

These are just my thoughts. Let me know what you think.

ahpohl, 2020-09-05:
Battery level is now reported in volt by default. Use the following lambda to report in percent instead:
```
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:4E:16:78"
    temperature:
      name: "ATC Temperature"
    humidity:
      name: "ATC Humidity"
    battery_level:
      name: "ATC Battery-Level"
      filters:
      - lambda: |-
          double res = (x - 2.2) / 9e-3;
          if (res > 100.0)
            return 100.0;
          else if (res < 0.0)
            return 0.0;
          else
            return res;

--- continue with config ---
```

benek984, 2020-09-05:
Hi,
How to add this new platform to my ESPHome installed as Add-on on HassOS?

Hass.io Add-on: ESPHome (dev)
Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: HassOS 4.12  (amd64 / qemux86-64)
 Home Assistant Core: 0.114.4
 Home Assistant Supervisor: 235

michaeldvinci, 2020-09-05:
> 1. (not yet officially released)  `atc_mithermometer`

@ahpohl Where can one get this to test out? Is it a custom component for ESPHome or for Home Assistant?

ahpohl, 2020-09-05:
Don't know about Hass.io. The instructions [above](https://github.com/esphome/feature-requests/issues/877#issuecomment-687568577) are for any full Linux distribution and a normal PC/Laptop

ahpohl, 2020-09-05:
Support in Hass.io is not there yet. The new component needs to merged with esphome/dev first and then in the next release it will be there also in Hass.io

ahpohl, 2020-09-05:
Browser compatibility for the Telink flasher is an issue though. Apparently the bluetooth feature is not supported by all browsers on all platforms. Nothing @atc1441 can do about it though. It looks like Chrome on Windows or Android is your best bet. Linux, MacOS and iOS are not supported it seems.

[https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth#Browser_compatibility](https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth#Browser_compatibility)

[image]

atc1441, 2020-09-05:
I can confirm that MacOS does work. Was tested on it by a friend.

danielkucera, 2020-09-08:
> @danielkucera As the method how to generate a new bindkey and upload to the device is now known (thanks to the marvelous work of @atc1441), did you meanwhile update your python script?

no, I have so many other things in progress so this is very low in my priorities. but feel free, it's opensource :)

ahpohl, 2020-09-08:
Actually with the Telink flasher working so well I do not see any need for further development. You have so many options now to use LYWSD03MMC sensors with ESPHome. Either use the stock firmware after activating the sensor with the Telink flasher and inserting a proper bindkey into the ESPHome config (`xiaomi_lywsd03mmc`), or flash the custom firmware without the need for a bindkey (apart from a fake one just to compile the firmware, and don't forget to switch to "Mi like" advertising). The third option would be to use the custom advertising with [ATC MiThermometer](https://github.com/esphome/feature-requests/issues/877#issuecomment-687568577). I have currently no plans to submit a PR for this component as it is not needed anymore.

nikito7, 2020-09-08:
I use xiaomi_cgg1 platform

Maybe "Mi Like" should be default option

atc1441, 2020-09-08:
@nikito7 sorry but will leave it on custom by default for now. Hope the custom will be submited some day :)

kquinsland, 2020-09-08:
A huge thanks to @atc1441 for your work in revering and crating a custom file. The 'no-solder' approach is a brilliant use of web-BT apis. I love the additions: showing the significant bytes of the BT mac on boot and altternating between showing battery % and humidity is a brilliant touch.

I don't want to spam, but if you look on Banggood, you can currently pick up a 3 pack of these guys for about $12 + the usual few $ for shipping from china.

I am eagerly awaiting adoption into ESPHome :).

ahpohl, 2020-09-09:
> I am eagerly awaiting adoption into ESPHome :).

No need to wait. Either use the `atc_mithermometer` component from my forked esphome repo with default custom advertising or switch to "Mi like" advertising and configure the sensor with already supported `xiaomi_lwsd03mmc` from the esphome dev branch. Both approaches are equivalent in terms of the custom firmware functionality. If you don't want to flash the custom firmware and keep the stock firmware, then use the Telink flasher "do activation" button to generate a new bindkey. The choice is yours.

I am going to prepare a PR against esphome-docs to add the documentation how to use custom firmware with ESPHome.

ahpohl, 2020-09-09:
@atc1441 Do you want to see support for the custom advertisement is ESPHome? The code is already there and just need to be prepared into a PR against esphome-dev.

atc1441, 2020-09-09:
@ahpohl yes would love to have it supported.

There is the new Xiaomi E-Ink C401 Thermometer upcomming with a port of the custom firmware, just as aditional info https://github.com/atc1441/ATC_MiThermometer/issues/16#event-3744484393

ahpohl, 2020-09-20:
Draft PR: [Add support for ATC_MiThermometer #1291](https://github.com/esphome/esphome/pull/1291) created. Todo: update esphome-docs documentation

ahpohl, 2020-09-20:
PRs https://github.com/esphome/esphome/pull/1291 and documentation updates https://github.com/esphome/esphome-docs/pull/777 created.

closed 2020-11-03 by jesserockz

53845714nF, 2020-11-23:
Hello great job, thanks.  
I have a small question. In which version is it in Home Assistant?
My system says:  Platform not found: 'sensor.atc_mithermometer'
I use Home Assistant 0.118.2. with esphome.

With kind regards
Sebastian

nikito7, 2020-11-23:
esphome dev

53845714nF, 2020-11-26:
Okay, thank you very much. 
I expected that. It was a little hard to find the URL for the repositorie (https://github.com/esphome/hassio).
But now building ESP Home has worked.

With kind regards

t-solak, 2021-02-11:
Thanks, @ahpohl for your great job to get the atc1441 firmware/format supported in esphome. Is working like a charm.

However, is there any chance to get the [pvvx](https://github.com/pvvx/ATC_MiThermometer) custom ADV format supported as well as in here, temp. and humidity are advertised more granular (x 0.01°C/x 0.01% vs. x 0.1°C/x 1%)?

I know that it doesn't really make sense from a precision point of view as the sensors specs got a +- 0,2°C variation. However, it would give me a trend and a cleaner graph in HA. :)

Would really be appreciated.

Regards, Talat.

pasiz, 2021-02-19:
> Thanks, @ahpohl for your great job to get the atc1441 firmware/format supported in esphome. Is working like a charm.
>
> However, is there any chance to get the [...]

You can find pvvx mithermometer in: https://github.com/pasiz/esphome/tree/dev/esphome/components/pvvx_mithermometer

t-solak, 2021-02-20:
> 
> You can find pvvx mithermometer in: https://github.com/pasiz/esphome/tree/dev/esphome/components/pvvx_mithermometer
>
@pasiz, I really appreciate your quick support and thank you for your fantastic work!
Running it on two ESP32 collecting 7 sensors in total - absolutely brilliant.

Cheers, T.

pasiz, 2021-02-20:
Thank you. Would be awesome if someone have time to write documentation of the new sensor type. I like also silky smooth graphing. I have 8 sensors running with 1 esp32 in my setup. That request just get me some drive to transfer my custom component to PR.

Regards,
Pasi

theCroh, 2021-04-14:
> You can find pvvx mithermometer in: https://github.com/pasiz/esphome/tree/dev/esphome/components/pvvx_mithermometer

Can you somehow use this on a HASS.IO installation? Is it enough to copy the files somewhere or do I need to do something else as well?

Peixeiro, 2021-04-25:
Hi, 

There is any way to set time on LYWSD03MMC from EspHome?
I set ATC to show time, but it add about 1 minute per day. If it is possible to set time from EspHome, the clock always show correct values.