esphome/feature-requests #878: update_interval for binary_sensor

dashdrum, 2020-09-02:
**Describe the problem you have/What new integration you would like**

Use the update_interval setting to cause a binary_sensor to send the state on the specified interval

**Please describe your use case for this integration and alternatives you've tried:**

When coding my motion sensors using C++, I included the function to send a state every 60 seconds.  Then, in Home Assistant, I would expire the state after a few minutes.  This would show if the sensor was no longer communicating with the host, and help show if wires had been cut or disconnected.  Using an ESPHome based sensor, I have to remove the expire_after setting in Hass or the sensor will show as unavailable when there is no movement over a time longer that the expire interval.  

An update_interval would serve as a heartbeat for the sensor hardware.

**Additional context**

glmnet, 2020-09-04:
Is the communication times out with the api the sensor goes to unavailable state. 

Are you talking Mqtt or esphome api?

dashdrum, 2020-12-04:
glmnet, I am talking MQTT.

Since submitting this feature request, I have learned more about the availability features of MQTT, and in an ESPHome/Home Assistant environment it works well.  However, other implementations of MQTT don't necessarily include that functionality. Having an update_interval or heartbeat option would allow the user to resend the status at a predetermined interval.

hansonap, 2024-03-03:
Agreed, I also have a use case for sending the value of a binary sensor every update_interval over MQTT. I guess my question is what determines when a component supports update_interval? Components derived from Sensor seem to support it but Sensor itself does not (at least according to the documentation).

It seems like a work around is to use the interval component.

glmnet, 2024-03-11:
The components that derives from PollingComponent supports that. I guess that is in the h file doc