esphome/feature-requests #879: Support min/max_value inputs for core Light component

ryanjohnsontv, 2020-09-03:
**Describe the problem you have/What new integration you would like**
Currently when using my PWM controlled RGBWW lights, the lights turn off at ~11% brightness within Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**
It would be nice to have the min_value and max_value fields that are available for Tuya dimmers apply to the core light component so that the frontend control of the lights is simplified. I know the brightness is calculated via PWM frequency and gamma correction, however with the current component I am unable to find a combination that scales as expected.

**Additional context**
FWIW, when using Tasmota I had no issue with the brightness percent being represented accurately, so I don't believe this to be a hardware limitation.

ElksInNC, 2021-02-24:
Have you tried adding gamma_correct: 1.0 to your light setup?  I was having the same problem on a single monochromatic backlight LED on a light switch and when I added that to my setup, the max and min values were honored when I moved the slider from 100 down to 0 in Home Assistant.  FWIW (I know this is an old post.)

oxan, 2021-06-04:
See also #920.

WeekendWarrior1, 2021-07-23:
min + max settings are a part of the float_output component that your RGBWW light uses for each of it's pins so:

```yaml

output:
  - platform: esp8266_pwm
    id: red
    pin: D3
    min_power: 0.11

  - platform: esp8266_pwm
    id: green
    pin: D4
    min_power: 0.11

  - platform: esp8266_pwm
    id: blue
    pin: D5
    min_power: 0.11

  - platform: esp8266_pwm
    id: white
    pin: D6
    min_power: 0.11

```

Should fix your issue (though this feature would still makes sense in non-PWM light components)

closed 2022-07-06 by nagyrobi