esphome/feature-requests #880: Add uart_id and energy on Pzemdc

lucasimons, 2020-09-03:
Now the pzemdc as, pzem 017, not support uart_id and energy meter, indeed the pzemac support all this two thinghs,
I would use two pzemdc to monitor the energy from the solar panel to battery, in the same board..
thanks for the support

 DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

randybb, 2020-09-04:
Physical layer is UART, but on top of it it is MODBUS. That means if you want to attach it to a different UART, then it is done under _modbus:_ componet like this: 
```yaml
uart:
  - id: uart_2
    rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 9600
    stop_bits: 1
modbus:
  uart_id: uart_2
  id: modbus_2 # needed only for multiple buses

sensor:
  - platform: pzemdc
    modbus_id: modbus_2
    current:
      name: "PZEM-003 Current"
    voltage:
      name: "PZEM-003 Voltage"
    power:
      name: "PZEM-003 Power"
    update_interval: 60s
```

lucasimons, 2020-09-04:
yes but if i want to use two pzem i have sone problem with the sensor

uart:
- id: uartbatteria
  rx_pin: D6
  tx_pin: D5 
  baud_rate: 9600
  stop_bits: 2
- id: uartpannello  
  rx_pin: D2
  tx_pin: D1
  baud_rate: 9600
  stop_bits: 2
  
  
modbus:
- uart_id: uartbatteria
  id: modbus_1 # needed only for multiple buses
- uart_id: uartpannello
  id: modbus_2 # needed only for multiple buses 
  
sensor:

  - platform: pzemdc
    - modbus_id: modbus_1
      current:
        name: "Batteria Corrente" 
      voltage:
        name: "Batteria Tensione"
      power:
        name: "Batteria Potenza"
      update_interval: 10s  
    - modbus_id: modbus_2
      current:
        name: "Pannello Corrente" 
      voltage:
        name: "Pannello Tensione"
      power:
        name: "Pannello Potenza"  
      update_interval: 10s

or like this

sensor:

  - platform: pzemdc
      current:
        name: "Batteria Corrente" 
      voltage:
        name: "Batteria Tensione"
      power:
        name: "Batteria Potenza"
      update_interval: 10s  
      modbus_id: modbus_1
      
  - platform: pzemdc  
      current:
        name: "Pannello Corrente" 
      voltage:
        name: "Pannello Tensione"
      power:
        name: "Pannello Potenza"  
      update_interval: 10s
      modbus_id: modbus_2

   but not work thanks for the support @randybb

napieraj, 2020-10-01:
closing issue

closed 2020-10-01 by napieraj