esphome/feature-requests #882: View and edit other files, e.g. !include files

ptr727, 2020-09-05:
**Describe the problem you have/What new integration you would like**

I am creating device templates, and using `<<: !include foo.yaml` such that only unique configuration needs to be done per device.
I would like the ability to edit any file in my configuration folder.

Alternatively a more complex mechanism of UI awareness of templates or includes, such that the UI knows what other files are being referenced, and allows for their editing.

**Please describe your use case for this integration and alternatives you've tried:**

As is, I need to use an external editor to edit any files that are not secrets or not the device config file.

**Additional context**

FreelancerJay, 2020-11-05:
Maybe an option to create not just new nodes, but to also create “common” files managed in the ESPHome UI?

dashdrum, 2022-06-22:
I support this request.