esphome/feature-requests #884: Support local control from Google Home

Joshfindit, 2020-09-07:
**Describe the problem you have/What new integration you would like**

Be able to control the ESPHome device from Google Home on the local LAN and *without* any proxies or go-betweens.

**Please describe your use case for this integration and alternatives you've tried:**

“Hey Google, ...
- ...turn on the light”
- ...tell me what the sensor value is”
- ...is \<BLE device\> visible?”
- ...set AC to 72 degrees”
- etc

**Additional context**

Google this year released [“local fulfillment”](https://developers.google.com/assistant/smarthome/concepts/local#understand_how_it_works) 
Now, smart devices are able to communicate directly with custom code running on the google home/nest devices.

alexbk66, 2025-03-23:
No comments at all?

ssieb, 2025-03-24:
Looks like you need to make some sort of app for Google Home.
I doubt that any of the main developers would be interested in this, so you'll have to find someone or do it yourself.