esphome/feature-requests #888: queue sensor data to use deep sleep mode

pille, 2020-09-08:
**Describe the problem you have/What new integration you would like**

there are timing issues when using deep sleep with a sensor, as the data will be retrieved by HA, which fails, when it's sleeping. when a sensor is updates once a minute, it may sleep the rest of the time, but we can't use this feature, as the ESP may not be queried, when awake.

the device should queue values (with timestamps) so that all values may be transmitted back to HA, on a successful connect.

**Please describe your use case for this integration and alternatives you've tried:**

sometimes the sensor is not reachable, when HA queries the device (e.g. network problems). the sensor still records data all the time, so nothing gets lost, but delivered late.

**Additional context**

a draw-back would be that HA can't act in real-time on sensor-data, but that's a trade-off with deep-sleep and sampling.

using a `FIFO` instead of a queue, may provide enough data locally on the device, that a graph can be displayed with the webserver feature (https://esphome.io/components/web_server.html).

pille, 2020-09-30:
this is a more verbose duplicate of #733.

napieraj, 2020-09-30:
closing as this is a duplicate, if you have any additional comments please append those to #733 or create a new issue

closed 2020-09-30 by napieraj