esphome/feature-requests #890: Make Device Online/Offline Indicator More Color Blind Friendly.

invinciberry, 2020-09-10:
**Describe the problem you have/What new integration you would like**

With the new 1.15 beta's WebUI redesign, the device's online/offline indicator was changed from a circular dot into a thin line on top of the device card. It became incredibly difficult to perceive it's colour for people with deuteranopia (red-green colour blindness). Please make it thicker or give us the ability to change its colour.

**Please describe your use case for this integration and alternatives you've tried:**

I would love to see which of my device is offline again with a simple glance. Not I have to rely on a Chrome colour picker add-on to determine the colour of the thin line.

**Additional context**

fensoft, 2020-09-15:
FYI I'm using dark reader chrome/firefox extension so with a dark background i'ts better.

Sebclem, 2021-01-26:
Any new on this ? 
It's just impossible for me to see if device is online/offline...

balloob, 2021-07-20:
Fixed in https://github.com/esphome/dashboard/pull/44. Will be released on Wednesday 👍

closed 2021-07-20 by balloob