esphome/feature-requests #891: Implement "pages" feature for MAX7219 Digit Display

nordeep, 2020-09-10:
**Describe the problem you have/What new integration you would like**

Is it possible to realize "pages" feature for MAX7219 Digit Display?
**Please describe your use case for this integration and alternatives you've tried:**

It would be useful for some "pixel animation".
**Additional context**

ill-tempered, 2020-12-28:
Would also be useful to show multiple information:

Eg: Time, weather, back to time, other sensors, back to time, RSS

olealm, 2022-05-16:
I am in the process of replacing firmware in all proprietary (and Tasmota) MQTT devices, and was "shocked" to see pages is not implemented for MAX7219 display in ESPHOME. I am using pages for all ESPHOME units that has a OLED 1306 display.
For the MAX7219 units I now have an /15 sec. automation with an "interval" updating the text sent to the units. Would very like to have this done locally in the ESPHOME unit, since the unit itself knows what time it is and whatnot.

TombaDude, 2022-05-30:
Yessss, we definitely need this function! Would make life just that much easier :)

nagyrobi, 2022-07-06:
Duplicate of https://github.com/esphome/feature-requests/issues/1480
(not pages concept for all displays, not just this one)

closed 2022-07-06 by nagyrobi