esphome/feature-requests #892: Please, add 2.9in(b) to Waveshare E-Paper Display supported models

regystro, 2020-09-11:
Hi.

I'm aware of https://github.com/esphome/feature-requests/issues/239, but since 2.9in is a smaller size and what I'm asking is already implemented in the [Arduino Waveshare E-Paper library by Yehui (@soonuse)](https://github.com/soonuse/epd-library-arduino/tree/master/2.9inch_e-paper_b), I don't think It'd be a matter of memory consumption, even for 3 colors.

Would it be easy to add support to this 2.9in r/b/w display beauty to the model list? I tried `2.90in` but it doesn't work.

Thank you so much.

nuttytree, 2021-02-11:
@regystro I think I can add this for you but I don't have one so I will need help with testing.  Can you tell me if the controller needs to be set to display type A or B for these?

regystro, 2021-02-11:
Hi @nuttytree. Thanks for dedicating time. I'm glad to help testing.
This is the display I have: https://www.waveshare.com/wiki/2.9inch_e-Paper_Module_(B) = Red / Black / White version
and needs to be set to type B: https://www.waveshare.com/wiki/E-Paper_Driver_HAT#Switch_settings.

nuttytree, 2021-02-20:
@regystro I went to work on this and discovered that the 2.9in B is already built (but with B/W support only).  To use it the model should be set to 2.90in-b.  Let me know if this works.  If it does not I can look into trying to fix it.  Adding the support for red is probably beyond my abilities (C++ is a relatively new language for me and my Python is not great either).

regystro, 2021-02-23:
Yep, thanks for pointing it out. B/W support was added and released just [two days](https://esphome.io/changelog/v1.15.0.html) after I filled this issue. I tested it the same day and it's working right.

I was aiming to get the 3rd color working, and like in your case, C++ is relatively new to me and my programming skills are way too rusty. Anyway I tried to rewrite some code and managed to replace the black pixels by red ones, but in a very hacky way. For the third color, another display buffer is needed (hence more memory) or find a way to reuse the B/W one in a tricky way.

Thanks again @nuttytree. I guess we'll have to wait for [color support in display base abstraction](https://github.com/esphome/esphome/pull/663#discussion_r302421163) and keep going from then on.

nuttytree, 2021-03-02:
@OttoWinter the 2.9in B display has already been added, this can be closed.