esphome/feature-requests #893: Implement animation for display components

nordeep, 2020-09-13:
**Describe the problem you have/What new integration you would like**

It would be great to have support for animation on the display.

**Please describe your use case for this integration and alternatives you've tried:**

The animation could give more dynamic to show data on display. Such as sun shine or rain or clouds for display weather conditions. Or say animated warning sign in some circumstances, or something like blinking reminders or whatever you want.
For now we may add a lot of images like frames of animation and after do like:
- by time.on_time trigger periodical change frame to show
- use switch case construction to show necessary frame

Not so intuitive and simple.
**Additional context**

It seems not to so hard to implement base animation. We already have python pillow library which support of GIFs images. We could parse GIF image, extract frames to array and use construction like:
it.animate(x, y, id(my_animated_gif), time_to_show_of_one_frame(or FPS));

PS: community can add animated GIFs, if we have the opportunity

Syndlex, 2020-11-11:
Hey,

I just looked into the code. I would be willing to contribute here.
But I have a question about the architecture here.
This could be added as an enum to the image class or as a child class of image. **What is preferred from the component Owner?**

The second question where I could not find an elegant solution was: **how we should update to the next image?**
Should this be done from the display component itself, from the image/animation component or from an automation of the User. 

I would prefer from the User Automation because the update freq can be dependent on the Display implementation.

@napieraj  are you the code owner here? did not found a owner here: https://github.com/esphome/esphome/blob/dev/CODEOWNERS

napieraj, 2020-11-11:
@Syndlex I have not written any display code