esphome/feature-requests #895: Include u8g2 lib into esphome

jfeldmaier, 2020-09-13:
**Describe the problem you have/What new integration you would like**

I want to add a UC1610 graphical display I formerly used in an Arduino project into my esphome project (mqtt home asistant display). The UC1610 is not supported by esphome but there is the u8g2 lib (https://github.com/olikraus/u8g2) available which supports a lot of graphical displays.

**Please describe your use case for this integration and alternatives you've tried:**

I have used the display in former Arduino projects together with the u8g2 lib - that was really handy. I tried to add a custom i2c component to control the display within esphome - but it was too complex for me.

**Additional context**

JZ-SmartThings, 2020-09-28:
+1 for this. Using Heltec Wifi Kit 32 with the built in OLED and having a hard time w/o usage of this library. My display is SSD1306 which a very common 128x64 OLED.

randybb, 2020-09-28:
@JZ-SmartThings Why? [SSD1306](https://esphome.io/components/display/ssd1306.html) is supported for ages.

JZ-SmartThings, 2020-09-28:
Hi @randybb I saw that page you linked and tried for awhile but could not get it to work. I believe that the Heltec Wifi Kit 32 uses software based i2c for the SSD1306 mounted to it. See the C++ line below and notice the SW piece. I know my GPIO pins are correct, etc. but even with i2c scanner, the Heltec cannot find the OLED (for me to know the correct hex address of the device). Hence the reason why I believe that the software based i2c is what's messing me up in making this work in ESPHome:

U8G2_SSD1306_128X64_NONAME_F_**SW**_I2C u8g2(U8G2_R0, /* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);      // Heltec Wifi Kit 32 Clock=15 Data=4 Reset=16 --- Wemos LoLin Clock=4 Data=5 Reset=16

randybb, 2020-09-28:
You need to pull pin 16 HIGH first that will enable this display and then you will see it in I2C scanner.

JZ-SmartThings, 2020-09-29:
@randybb I appreciate the suggestion! It truly moved me forward for sure... not sure why the code did not work before but I've stripped it down. Your recommendation helped because I saw that when I pull GPIO16 high, the i2c scanner finally saw the 0x3C address of the OLED. In the end though, I didn't end up needing it and you were 100% correct that SSD1306 works like a charm with ESPHome. Thanks again!

For folks who use Heltec Wifi Kit 32 (ESP32) in need of reference, the below is bare minimum for enabling the OLED... and you do need to take arial.ttf form c:\windows\fonts (drag/copy Arial to your desktop) and put it in the ESPHome folder per the instructions:

```
esphome:
  name: esphometestnode
  platform: ESP32
  board: heltec_wifi_kit_32

wifi:
  ssid: "SSID"
  password: "wifipassword"
  domain: ".local"
  power_save_mode: none

  ap:
    ssid: "Esphometestnode Fallback Hotspot"
    password: "hotspotfallback"

logger:
api:
ota:

i2c:
  - id: bus_b
    sda: 4
    scl: 15
    scan: True

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: 16
    address: 0x3C
    lambda: |-
      it.print(6, 6, id(font1), "Hello World!");

font:
  - file: "arial.ttf"
    id: font1
    size: 12
```

JZ-SmartThings, 2020-09-29:
@randybb actually there is a bug (or limitation)... I'm using temperature/humidity/pressure sensor BME280 and it lives on a different i2c bus. Whenever whenever I would try to use TWO i2c bus definitions (like in the docs), the OLED will NOT work and cause this this error: **[22:21:08][E][ssd1306_i2c:031]: Communication with SSD1306 failed!**

I found the solution for my issue... the key is to use very obscurely documented key value pair in YAML under the display and likely under the BME config. I found **i2c_id** mentioned only on [two pages in the documentation](https://esphome.io/search.html?q=i2c_id) and it made all the difference. After using the below config for the display it worked like a charm notice the i2c_id toward the bottom... thanks again for your help @randybb 

```
### OLED & BME280
i2c:
  - id: bus_a
    sda: 21
    scl: 22
    scan: True
  - id: bus_b
    sda: 4
    scl: 15
    scan: True
### OLED & BME280 --- END

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: 16
    i2c_id: bus_b
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(arial12), "${device_name}");
```

closed 2022-07-01 by nagyrobi