esphome/feature-requests #896: Support for Samsung A/Cs via IR Remote Climate

xorbital, 2020-09-14:
**Describe the problem you have/What new integration you would like**

I would like to integrate my Samsung A/Cs into Home-Assistant, the new IR Remote Climate seems like a great fit for that.
This is the remote:
[image]

**Please describe your use case for this integration and alternatives you've tried:**

The A/Cs used to have an open port for local API access, however they only use the proprietary Smartthings cloud now.

**Additional context**

There are some advanced commands the remote can send, like a "windfree" mode, which closes the flap, Eco Mode, Quiet Mode, Timers and a "Fast" mode. On my unit (AR12TXEAAWKNEU) some of the features of the remote (like direct/indirect, purify) aren't suported, but they are still present on the remote.

I captured some IR codes and tried to decode it as far as I can, please let me know what I can do now :)

This is the command for Cooling, 16°C, Windfree (Fan Auto):
`02,92,0F,00,00,00,F0,01,D2,FE,7B,00,11,F0`
I got this with the https://github.com/ToniA/arduino-heatpumpir/ library. 
What I found out so far:

```
02 - ??
92
  9 - Quiet mode: 8: quiet / special, 9: normal
  2 - ?
0F - ??
00
  0 - special mode: [0: normal, 1: indirect, 2: direct]
  0 - ??
00 - ??
00
  0 - Beep, 0: off, 2: on
  0 - ?
F0 - ??
01 - ??
D2 - Checksum 
FE
  F - Swing: A: vertical, B: horizontal, C: vertical+horizontal, F: off
  E - Not sure; F:swing, E: no swing, also E: 17-30°C, F: 16°C
7B
  7 - options? 7: normal, 6: Display off, E: Clean
  B - B: Windfree, 1: normal, 7: Fast, F: ECO
00
  0 - temperature (0: 16°C bis E: 30°C), always 8 in Fan Mode, when selecting "Fast" the last set temperature is sent
  0 - ?? 0: (normal), [1: purify] 2: clean
11
  1 - Mode; 1: Cool, 2: Dry, 3: Fan, 4: Heat, 8: Auto
  1 - Fanspeed; 1: Auto, 5: Low, 9: Medium, B: High, F: Turbo
F0 - ??
```
When pressing the power button, a longer command is sent including timers, haven't checked that out yet.
Also, some combinations are invalid, e.g. you can't use the highest fan setting in quiet or windfree mode (auto only).
Other observations:
CLEAN is a one-time command and gets saved in the A/C, not the remove (Clean mode basically blows air through the unit after cooling so you'll get rid of condensation)
Purify, direct and indirect have no function on the AR**TXFC***/AR***TXEA units.

Special commands:
01,72,0F,00,00,70,83: Show Usage
(There are more, like reset filter usage I think)

Fusseldieb, 2020-09-25:
I have the same AC and would love to control it!

Currently using a simple "switch" to send an IR command for on and another for off

xorbital, 2020-09-27:
I'm currently trying to use IRremoteESP8266 for that: https://github.com/crankyoldgit/IRremoteESP8266/issues/1277
It might be possible to use that code for esphome than as well...

Here are the signals I've decoded so far (short codes): https://cryptpad.fr/sheet/#/2/sheet/view/REURkHbGuQdU+wQkBcwKBXmccftjIB+Tyzin34EXs-g/
I got this decoded with https://github.com/ToniA/Raw-IR-decoder-for-Arduino
However if you don't have a shitty IR decoder like me you might be able to use the examples from IRremoteESP8266 as well...

iglu-sebastian, 2021-01-22:
https://github.com/esphome/feature-requests/issues/444 has some more info

xorbital, 2021-11-06:
The unit is now supported in the master branch in the upstream library and will make it's way into ESPHome eventually.

closed 2021-11-06 by xorbital