esphome/feature-requests #903: Add support for play audio

blademckain, 2020-09-19:
**Describe the problem you have/What new integration you would like**
support for playing audio files

**Please describe your use case for this integration and alternatives you've tried:**
connecting a speaker to an esp32, audio file could be played

**Additional context**
an example here
[https://community.home-assistant.io/t/turn-an-esp8266-wemosd1mini-into-an-audio-notifier-for-home-assistant-play-mp3-tts-rttl/211499](https://community.home-assistant.io/t/turn-an-esp8266-wemosd1mini-into-an-audio-notifier-for-home-assistant-play-mp3-tts-rttl/211499)

docteurzoidberg, 2020-09-29:
would be great to implement audio !

docteurzoidberg, 2020-11-22:
ESP8266Audio and ESP8266 SAM (tts) would be great, if it can fit !
https://github.com/earlephilhower/ESP8266Audio
https://github.com/earlephilhower/ESP8266SAM

Dodoooh, 2021-09-08:
is there any news here? 

It is possible: https://www.instructables.com/MQTT-Audio-Notifier-for-ESP8266-Play-MP3-TTS-RTTL/

But I see even greater possibilities with ESPHome.

escoand, 2022-03-30:
Found this nice project https://github.com/schmurtzm/MrDiy-Audio-Notifier
Maybe it could be used somehow.

nagyrobi, 2022-07-01:
Implemented: https://esphome.io/components/media_player/i2s_audio.html

closed 2022-07-01 by nagyrobi