esphome/feature-requests #904: Automation for start of OTA updates

DeanCording, 2020-09-19:
**Describe the problem you have/What new integration you would like**

An automation at the start of the OTA update process so that connect sensors and devices can be put into a safe mode whilst the upgrade is in progress.

**Please describe your use case for this integration and alternatives you've tried:**

OTA upgrades suspend processing of automations and sensors whilst the upgrade is in progress.
This can lead to unsafe states such as a heating element being left on or a motor driving past a stop where these devices are being monitored and managed with ESPHome.

In addition, using the native ESP8266 SPISlave library will cause a crash if an SPI message is received during an OTA upgrade due to a clash with the SPI writes to flash.

Having an automation triggered at the start of an OTA upgrade would allow the connected devices to be rendered safe or the SPISlave interrupt be disabled before the upgrade process takes control.

**Additional context**

Restart and Shutdown switches currently have this functionality.

mmakaay, 2021-04-28:
I created a pull request for this feature:
https://github.com/esphome/esphome/pull/1714

nagyrobi, 2022-07-07:
Implemented: https://esphome.io/components/ota.html#ota-automation