esphome/feature-requests #910: Add RGBWW E1.31 Support

mdeluca1992, 2020-09-21:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jdubz666, 2020-09-21:
Yes please!!

napieraj, 2020-09-30:
there is a custom_component that offers E1.31 support:
https://github.com/ayufan/esphome-components#23-e131

feel free to try it out and offer the author feedback, maybe we can get it merged

jdubz666, 2020-10-01:
> **Describe the problem you have/What new integration you would like**
> 
> Ability to control a RGBWW light using E1.31.
> 
> **Please describe your use case for this integration and alternatives you've tried:**
> 
> Unable to add E1.31 effect to RGBWW lights.
> 
> **Additional context**
> 
> [image]

Wow thank you! I'll give it a go and let you know!

jdubz666, 2020-10-01:
Unfortunately this component is only for the LED strips. I'm hoping this will be available on the RGBWW platform.

There is already an open issue about it on the component. See link.

https://github.com/ayufan/esphome-components/issues/14

Please let me know if you have any other ideas to get E1.31 on RGBWW platform.

j2mc, 2020-10-08:
I'd love this as well, but to add on I would love E1.31 control for regular lights or relays as well if that's a possibility.

Also as a workaround I've setup the RGB portion of my bulbs using fastled(they have the SM16716 chip) so at least I do get RGB E1.31 control of that right now!

jdubz666, 2020-10-08:
Thanks for the tip! @j2mc. I'll give fastled a shot! Sounds like a good work around. Mine use SM16716 chip as well.

jdubz666, 2020-10-08:
> I'd love this as well, but to add on I would love E1.31 control for regular lights or relays as well if that's a possibility.
> 
> Also as a workaround I've setup the RGB portion of my bulbs using fastled(they have the SM16716 chip) so at least I do get RGB E1.31 control of that right now!

Unreal it worked! :D

WholeV92, 2020-11-27:
Bumping this as I failed with it this morning. I found an alternative code to use but would really want to keep it in ESPHome.

[https://github.com/thehookup/E131-MQTT-RGBW-Lights/blob/master/E131RGBW_MQTT_Lohas_CONFIGURE.ino](url)

jdubz666, 2020-11-27:
@WholeV92 I tried following that tutorial but ended up bricking a bulb lol... 

What's your issue? I can try helping or posting my code for using FastLED.

WholeV92, 2020-11-27:
@jdubz666 not really much of an issue, more of a want to be able to have E131 control in ESPHome.  I was able to get the bulbs flashed with the firmware from TheHookup above to work but its not a clean process.  I am not a fan of adding into HA yamls.  

BTW - if you are US based the Lohas bulbs can be had for 22$ for 4 of them today.

oxan, 2021-06-04:
See also #1107 and #966.

cfamfl2019, 2021-06-15:
I also have a few places where this would be very useful.  Perhaps @ayufan could make this happen?

tony-fav, 2021-12-03:
I have put together an example here for a five channel bulb controlled by either DDP or E131

https://github.com/tony-fav/esphome-sync/blob/main/ddp_e131/five-channel-bulb.yaml

ayufan, 2021-12-03:
Pretty hacky @tony-fav :)

fortl, 2021-12-09:
oh, my fault! I've just found partition light component mention above. It helps to do e131 in my cases

closed 2022-07-01 by nagyrobi