esphome/feature-requests #914: Use LEDC PWM for ST7789V's backlight_pin

GHF, 2020-09-27:
**Describe the problem you have/What new integration you would like**

I'd like to use LEDC instead of GPIO output to control the `backlight_pin` of the ST7789V LCD on my TTGO T-Display project. 

So instead of:
```
display:
  - platform: st7789v
    backlight_pin: 4
    cs_pin: 5
    dc_pin: 16
    reset_pin: 23
```

I'd like to have:
```
output:
  - platform: ledc
    pin: 4
    id: ledc_display_backlight

display:
  - platform: st7789v
    backlight_pin: ledc_display_backlight
    cs_pin: 5
    dc_pin: 16
    reset_pin: 23
```

**Please describe your use case for this integration and alternatives you've tried:**

I don't really know the parser and schema well enough to hack together the above syntax, so what I've done is use an unused GPIO for the `backlight_pin` key then independently control the LEDC for GPIO4 independently of the ST7789V integration: https://github.com/GHF/iaq-node/commit/2642ecc1a425c1950ed78629615e030c20a13342

Per the [schematic](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf), GPIO4 controls the backlight LEDs through Q0 so it can be PWMed extremely fast and this works great, but I think PWMed backlight control should really be a part of the integration.

**Additional context**

genebean, 2023-12-24:
This sounds like something I'd find useful as well

genebean, 2023-12-24:
FWIW, I was able to comment out the backlight pin in the display component, add a ledc output component, and control the ledc via a light component. This is working really well.

ssieb, 2023-12-24:
This has been possible probably always and certainly is now.  It's even documented in the docs for this display.  If you get an error about reusing pins, then you can set `backlight_pin: no` in the display config.  But I've also been told that you should use the [ILI9xxx](https://esphome.io/components/display/ili9xxx) component for this displayl now anyway.

closed 2023-12-24 by ssieb

GHF, 2023-12-26:
> This has been possible probably always and certainly is now. It's even documented in the docs for this display. If you get an error about reusing pins, then you can set `backlight_pin: no` in the display config. But I've also been told that you should use the [ILI9xxx](https://esphome.io/components/display/ili9xxx) component for this display now anyway.

To be clear, this was absolutely _not_ possible at the time I filed this (see the workaround in OP where I had to assign some other pin to `backlight_pin`) and _is_ possible now thanks to the authors of several key changes:
- https://github.com/esphome/esphome/pull/2180
- https://github.com/esphome/esphome/pull/5476
Notably, the latter made `backlight_pin: no` available to fix the regression introduced in 2023.9.0 (esphome/esphome#5337) that makes `model` default a pin into `backlight_pin` if omitted, requiring a fix such as: https://github.com/GHF/iaq-node/commit/4a066b5d5afbac9aab3ab9f69218fde50cf80553

All, I really appreciate the work on this in the years since I've been using ESPHome! Also thanks @ssieb for the pointer to ILI9xxx.