esphome/feature-requests #915: Status_LED - new option to turn the status LED ON/OFF as the default --- Currently Just OFF is Available

JZ-SmartThings, 2020-09-27:
**Describe the problem you have/What new integration you would like**

I'm using the **status_led** component and the only issue is that the LED is the status for Wifi per the sticker on the specific (KAB, WiOn or Eco Plugs) ESP8266 device I am messing around with. I simply would love to default the light to ON rather than this statement [in the docs ](https://esphome.io/components/status_led.html)that the status_led: **Stay off otherwise.** 

**Please describe your use case for this integration and alternatives you've tried:**

Basically would like to use the status LED definitions that match the stickers/markings on the physical switch. It would be great to be able to control whether the LED defaults to ON or OFF rather than just defaulting to off like currently being done with not proper/simple alternatives.

Here's the alternative I've tried but it does cause a log flood: https://community.home-assistant.io/t/esphome-esp32-wifi-connected-led/184399

**Additional context**

napieraj, 2020-10-03:
if you invert the pin then *on* becomes off and vice versa, can you post your status_led configuration?

JZ-SmartThings, 2020-10-03:
That's perfect, all I needed... thank you @napieraj! I keep forgetting that inverted is always available for GPIO pins.

closed 2020-10-03 by JZ-SmartThings