esphome/feature-requests #916: Support more ESPCAM camera modules

tdamsma, 2020-09-27:
**Describe the problem you have/What new integration you would like**
Currently (at least) the OV2640 camera module is supported in the esp32_camera component. In the espressif [esp32-camera repo](https://github.com/espressif/esp32-camera) also the following are supported: OV2640, OV3660, OV5640 and OV7725 
However I can't get my OV7725 to work as it does not support PIXFORMAT_JPEG as format, only GRAYSCALE,YUV422 and RGB565. However it does not have an ir filter so theov7725 could potentially be a nice diy security camera. 

**Please describe your use case for this integration and alternatives you've tried:**
I tried to change the PIXFORMAT but this immediately led to lots of errors. Espressif provides a frame2jpg function so someone with more cpp experience could probably fix this easily

**Additional context**

zagi988, 2022-09-19:
Any chance OV5640 working on ESPhome?

patstave, 2022-12-30:
I would also like to know if OV5640 can be handled on ESPHome

nagyrobi, 2023-03-19:
See https://github.com/esphome/esphome/pull/4580

Rudd-O, 2024-03-21:
@nagyrobi that PR does not address OP's request.

closed 2026-02-19 by bdraco