esphome/feature-requests #917: Support for the SPI RF remote reciever module (nRF905)

sebasptsch, 2020-09-28:
**Describe the problem you have/What new integration you would like**

Support for the nRF905 Transceiver module through SPI.  

**Please describe your use case for this integration and alternatives you've tried:**
Anyone buying this module will be able to plug-and-play using spi  and the remote component.

**Additional context**
Bought the module from Jaycar in Australia.
[Datasheet](https://www.jaycar.com.au/medias/sys_master/images/images/9426951372830/XC4522-dataSheetMain.pdf)
[Product Page](https://www.jaycar.com.au/arduino-compatible-rf-transceiver-module/p/XC4522)

sebasptsch, 2020-10-03:
[Arduino Example Code](https://www.electrodragon.com/w/NRF905)

sebasptsch, 2020-10-04:
[nRF905 Library](https://github.com/zkemble/nRF905)

closed 2020-10-22 by sebasptsch

renini, 2020-12-21:
@sebasptsch may i ask why you closed this. I was actually just looking for the same, did you get this module to work with esphome?

sebasptsch, 2020-12-21:
> @sebasptsch may i ask why you closed this. I was actually just looking for the same, did you get this module to work with esphome?

I gave up and purchased a different module on eBay which is proven to work. I figured it would be a lot of effort for not very many users trying to get that to work.

renini, 2020-12-21:
> > @sebasptsch may i ask why you closed this. I was actually just looking for the same, did you get this module to work with esphome?
> 
> I gave up and purchased a different module on eBay which is proven to work. I figured it would be a lot of effort for not very many users trying to get that to work.

Ah oke, what did you end up buying? Im using the esp32+nrf905 and using a arduino api webinterface https://github.com/eelcohn/nRF905-API. But figured esphome would give better integration...

sebasptsch, 2020-12-21:
I bought a standard receiver and transmitter set which work with esphome by default and integrate nicely.

xirixiz, 2021-02-21:
What exacly did you buy @sebasptsch? I`m using https://github.com/eelcohn/nRF905-API but it's very unstable.

sebasptsch, 2021-02-21:
> What exacly did you buy @sebasptsch? I`m using https://github.com/eelcohn/nRF905-API but it's very unstable.

I believe one of the first comments on this thread has all that information.