esphome/feature-requests #918: Thermostat: Remember temperature when switching to/from away mode

Sebclem, 2020-09-28:
**Describe the problem you have/What new integration you would like**

With the thermostat component: 
When to change the target temperature, switch to `away` mode and get back to `home` mode, the target temperature fall back to the `default_target_temperature_low` value. It will be interesting to remember the previous value and maybe add a setting to let the user choose if he want to fallback to default or remember the user changed value.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

wernight, 2021-05-20:
I actually noticed the same thing. Initially I implemented this myself using global variables that are stored and it worked, but now using the built-in thermostat it doesn't seem to recall any set value.

Here is my config for it:

```yaml
climate:
  - platform: thermostat
    id: my_thermostat
    name: "${devicename}"
    sensor: temperature
    default_target_temperature_low: 20 °C
    away_config:
      default_target_temperature_low: 5 °C
    heat_action:
      - switch.turn_on: floor_heating_switch
    idle_action:
      - switch.turn_off: floor_heating_switch
```

Once running, from Home Assistant:

 1. Set temperature to something else and wait a few seconds
 2. Change to Away mode for that thermostat
 3. Change back to Home mode for that thermostat

Expected: It restores the temperature set at step (1).
Actual: It restores the default temperature.

I'd expect that my device keeps a temperate set for home and one set for away. I get that some people might want to always go back to the hardcoded value in this scenario but that feel like the wrong behavior for most people. One point to back this up would be Nest Thermostats.

Alex9779, 2021-11-23:
As this is open for a while and I am looking forward to have a better control I hacked it into the component but I am not sure if I am missing something so I wanted to share to discuss. @kbx81 maybe you can have a look and comment...

I added three values to the `ThermostatClimateTargetConfig` struct to be able to store the setpoints:
```
struct ThermostatClimateTargetTempConfig {
 public:
  ThermostatClimateTargetTempConfig();
  ThermostatClimateTargetTempConfig(float default_temperature);
  ThermostatClimateTargetTempConfig(float default_temperature_low, float default_temperature_high);

  float default_temperature{NAN};
  float default_temperature_low{NAN};
  float default_temperature_high{NAN};
  float temperature{NAN};
  float temperature_low{NAN};
  float temperature_high{NAN};
  float cool_deadband_{NAN};
  float cool_overrun_{NAN};
  float heat_deadband_{NAN};
  float heat_overrun_{NAN};
};
```
and then used them in `change_away_` function:
```
void ThermostatClimate::change_away_(bool away) {
  if (!away) {
    if (this->supports_two_points_) {
      this->away_config_.temperature_low = this->target_temperature_low;
      this->away_config_.temperature_high = this->target_temperature_high;
      this->target_temperature_low = std::isnan(this->normal_config_.temperature_low) ? this->normal_config_.default_temperature_low : this->normal_config_.temperature_low;
      this->target_temperature_high = std::isnan(this->normal_config_.temperature_high) ? this->normal_config_.default_temperature_high : this->normal_config_.temperature_high;
    } else
      this->away_config_.temperature = this->target_temperature;
      this->target_temperature = std::isnan(this->normal_config_.temperature) ? this->normal_config_.default_temperature : this->normal_config_.temperature;
  } else {
    if (this->supports_two_points_) {
      this->normal_config_.temperature_low = this->target_temperature_low;
      this->normal_config_.temperature_high = this->target_temperature_high;
      this->target_temperature_low = std::isnan(this->away_config_.temperature_low) ? this->away_config_.default_temperature_low : this->away_config_.temperature_low;
      this->target_temperature_high = std::isnan(this->away_config_.temperature_high) ? this->away_config_.default_temperature_high : this->away_config_.temperature_high;
    } else
      this->normal_config_.temperature = this->target_temperature;
      this->target_temperature = std::isnan(this->away_config_.temperature) ? this->away_config_.default_temperature : this->away_config_.temperature;
  }
  this->preset = away ? climate::CLIMATE_PRESET_AWAY : climate::CLIMATE_PRESET_HOME;
}
```

This seems to work, the setpoints are reset to the previous values when switching modes. Though saving the inactive mode's setpoints accross reboots does not work as the device does not save the values and the restore function just restores the values of the last active mode. I don't think that is even possible because HA just saves and restores the values of the last state and the values of a different mode are not part of that state.

What came to my mind while working on this is that this home/away thing is just some special preset handling when looking at it from a `climate` point-of-view, isn't it? That was different some time ago, but "away" functions are deprecated now. Everything is now about presets, presets from `esphome::climate::ClimatePreset` and custom presets.
So I thought about also adding the possibility to add custom presets which might help me to solve another case I want to control in my house where switching just between home and away might not be enough.
So what do you think about that too?

Alex9779, 2021-11-24:
Ok creating a feature flag for saving temps is reasonable as always I just thought about my use case for the moment :P

I thought about 2. a while because initially I didn't really understand it, but I think I got what you mean by now. You're talking about an ESP device with a local user interface. As I don't have such things it wasn't clear to me at first but now I agree. But that seems not too hard if the component is more preset aware.
I don't like the hard coded "user preset" so, in combination with other components you need in such a device like a rotary encoder this could be handled by a new action for thermostat to set temperatures and switch preset before. The preset does not have to be hard coded in this case. It is then in the config but you have full control over it. You can name it, with the other components you can control when it gets activated with the new action. With another flag you can switch the preset to be just local so that remote changes first switch back to the previous profile for example before changing the values.

If I look at the old controllers of our floor heating there was a three state switch to switch manually between different presets, day, auto (timed) and night. For this to handle we need another action then to switch presets. As far as I can see there is no action to switch between home and away at the moment´, so implementing a switch on a device would not be easily possible.

Some questions on how to continue with this, do you have anything other than ideas and plans at the moment? Or do you have plans that you will get on this in the near future?
How are the chances for that quick hack (with a feature flag added) to get into the codebase for the moment?

kbx81, 2021-11-24:
I think what needs to happen before anything else is to solidify handling of presets and re-work the related part of the code. I was also planning to better separate out the "auto" and "heat_cool" modes as there is some confusion/entanglement of the two as it is now and this needs to be cleaned up.

I'm hesitant to introduce a "hack" for this as it means we bring something new into the config and then have to re-work it later which likely means a breaking change. As these things (arbitrary presets and storing/recalling modified values for them) are not-insignificant feature additions I think it should be well-planned so we're not introducing new things and then breaking them shortly thereafter.

As it relates to presets, I was thinking a bit more about it today and perhaps each preset could have a flag to indicate whether it is a "fixed" preset or an "adjustable" preset. This is probably cleaner approach and I think it moves closer to what you are describing, too. 😄

Alex9779, 2021-11-24:
Fine I am ok with what you say and I understand the problems that may come up if we just use that hack for now while there is work to do to generally rework some parts of the component...
For now and my use case I think I will use my hack for my devices so I can switch and solve some needs. Though I am looking forward for more than two profiles with stored set points to take my control needs another step forward.

The component is rather complex because of all the options and the inheritance from the climate component and use cases for which it is meant for so I might not be able to provide a decent suggestion on how to solve what we discussed.

But I offer my assistance with testing and concepts if needed. Don't hesitate to mention me if you work on something you want a second opinion...

closed 2022-07-01 by nagyrobi

nagyrobi, 2022-07-05:
I think this should be reported as an issue not a feature request then, because it targets a different attention level.

nagyrobi, 2022-07-05:
That PR looked to me like a solution targetting this request.

Alex9779, 2022-07-05:
Ok fine, well it targets some of the discussion between regarding presets. Seems it got disregarded when the new implementation was done, will start an issue for that, thanks...

nagyrobi, 2022-07-05:
As far as I understand the workflow:
- there was a feature request
- a solution was proposed in a PR which was merged => feature fulfilled (this issue should have been closed back then)
- feature requester not satisfied still => PR likely needs fixes => this is a defect/issue for the code already in, thus not a request for a new feature again.
- should open an issue, optionally mentioning this feature request in it.

MrMDavidson, 2022-07-05:
Given presets now (potentially) control all aspects of the thermostat (mode, fan, swing, temperature) would all of these get stored and restored when you move back and forth between presets? What happens when the preset doesn't specify a property and thus doesn't change it? (Eg. If a preset doesn't specify a fan mode then the fan mode doesn't get changed when you switch to the preset). Would it save the property anyway? Ignore it and keep which ever old state it got from the new property? Would this behaviour be configurable? At the thermostat level or preset level?

Alex9779, 2022-07-05:
Ya I agree, but I do not find any reference in the PR to this request. I was involved and just got notice that there is something going on with that component when you closed this request. So for me the maintainer and the contributor didn't pick up what this request was about and IMHO it should not have been closed...

nagyrobi, 2022-07-05:
When a dev starts to read this whole stuff will get very confused.
If you feel that the implemented PR is not about this request, then, for the sake of clarity, and the higher chances to be clearly understood and considered, I'd suggest to start fresh and open it again as a new feature request.

There were literally more than 1k so called "feature requests" here which in majority were either solved by alternative means, just reading the manual, implemented by custom components, PRs, or were duplicates etc. Need to brush things up and clean up stuff otherwise noone will use this list for anything.

Alex9779, 2022-07-05:
@MrMDavidson thanks for having a look, yeah there are some questions how this should work, if you have a look at what I discussed above then you see there have been some points, @kbx81 pointed a use case out where this is not wanted so we discussed about a feature flag to enable storing changes in a preset.
We talked about presets defining every aspect and restoring those values when selected and we discussed the need of some people, the author of this request and me for the moment, who like to have at least some values stored.

Since we have the presets now thanks to your work, a flag for each preset to store changes and not to load defaults maybe would be the easiest thing because as you can define as much as presets you like you can have fixed presets restoring always their defaults and other presets having defaults just for initial startups after resets or other cases where the device lost the stored values.

nagyrobi, 2022-07-05:
> how I need it to work

This described as precisely as possible in a new clean FR would be best, just mentioning the names of people you think are worth to be notified.