esphome/feature-requests #919: Support Sleeping for PMSX003 or warn about lack of sleep functionality.

raccettura, 2020-09-28:
**Describe the problem you have/What new integration you would like**

The PMSX003 sensors support a sleep mode. 

Example implementation can be seen here:
https://github.com/SwapBap/PMS

Switching to polling every (user defined) X minutes, waking for Y seconds (to let the sensor warm up) and polling data before setting it back to sleep would allow the sensor to last much longer and make it more practical to use with esphome.

**Please describe your use case for this integration and alternatives you've tried:**

The lifetime of the sensor thanks to a moving fan, and the eventuality of filling with dust is shortened if it's not put to sleep between reads.  

If not done this way, it should ideally be noted in the documentation that the current implementation will reduce the life expectancy of a somewhat expensive sensor.

**Additional context**

redstreet, 2020-11-15:
Started [this thread](https://community.home-assistant.io/t/ignoring-sensor-values-during-warm-up-after-periodic-wake-up/245401).

ingria, 2021-03-31:
Sleep cycle can easily be implemented via the **switch** platform and **interval**:

```yaml
switch:
  - platform: gpio
    pin: 
      number: SOME_PIN_ID
    id: pms_start
  
interval:
  - interval: 300s
    then:
      - switch.turn_on: pms_start
      - delay: 30s
      - switch.turn_off: pms_start
 ```

Like @redstreet said, the main problem here is that the sensor has some warmup period, during which the measurements must be discarded. So it should wake up -> wait 30s -> start emitting data -> go to sleep.

There's a hardware ~~solution~~ workaround: just use an external fan that would constantly pump the air through the sensor (like [this one](https://a.aliexpress.com/_mrfmOLv)).