esphome/feature-requests #920: minimum/maximum brightness on light

docteurzoidberg, 2020-09-29:
**Describe the problem you have/What new integration you would like**

I would like to be able to set the minimum, and maximum brightness in a light component. 

For the maximum brightness, I actually use color_correct: [50,50,50] in my config but a generic 'maximum brightness" option would be fine too.

 Nothings exist for the minimum,

**Please describe your use case for this integration and alternatives you've tried:**

Actually, when I set the maximum brightness to 50% on all channels with the color_correct option in a light component, my leds doesn't light on when set to under 20% brightness when turn_on. 

I would like to have a minimum 20% set in config so the light will be on when turned on

jmvaz, 2020-10-16:
I have the same problem. Just setup one arilux LC06 module, and above 7% the LEDs just turn off, I need to set up that the minimum allowed brightness is 7 %, otherwise, the leds will be shutdown.

gitolicious, 2020-11-30:
For my LED strip, values below 10% are "off", would be awesome to have this!

xuefer, 2021-01-07:
> For my LED strip, values below 10% are "off", would be awesome to have this! I could imagine something similar to `min_power` of the [Output Component](https://esphome.io/components/output/index.html#base-output-configuration).

same problem here. can't solve with esphome min_power since it remove 0 (off) output

RAYs3T, 2021-02-10:
Also running into this. Would be really nice to be able to set a minimum, because my strips start to flicker if they run under 14%

xuefer, 2021-02-10:
the only way is to patch min_power code to skip applying min_power on zero value

jmvaz, 2021-02-10:
> the only way is to patch min_power code to skip applying min_power on zero value

How can we do that? :)

xuefer, 2021-02-18:
[0001-allow-turn-off-output-even-for-non-zero-min_power.patch.gz](https://github.com/esphome/feature-requests/files/6000786/0001-allow-turn-off-output-even-for-non-zero-min_power.patch.gz)
```
From e5c060cd30f049a5f68b4b90ba9c2a6991a9b0c7 Mon Sep 17 00:00:00 2001
From: Xuefer <xuefer@gmail.com>
Date: Thu, 7 Jan 2021 22:26:26 +0800
Subject: [PATCH] allow turn off output even for non zero min_power

diff --git a/esphome/components/output/float_output.cpp b/esphome/components/output/float_output.cpp
index 0d536d0..ef4c787 100644
--- a/esphome/components/output/float_output.cpp
+++ b/esphome/components/output/float_output.cpp
@@ -30,7 +30,7 @@ void FloatOutput::set_level(float state) {
   }
 #endif
 
-  float adjusted_value = (state * (this->max_power_ - this->min_power_)) + this->min_power_;
+  float adjusted_value = state > 0.0f ? (state * (this->max_power_ - this->min_power_)) + this->min_power_ : 0;
   if (this->is_inverted())
     adjusted_value = 1.0f - adjusted_value;
   this->write_state(adjusted_value);
-- 
2.29.2
```

pmannk, 2021-03-02:
Ran into this issue myself tonight too. LED turns off < 10% and flickers under 20%.
Thanks for the proposed patch @xuefer !

WeekendWarrior1, 2021-07-23:
Should work without xuefer's patch now.

robertvorthman, 2021-07-23:
Still hoping the min_power parameter gets added to the Light component because my fastled_clockless platform lights flicker below 15%.  Xuefer's patch seems to only apply to PWM lights that use the Output component.

WeekendWarrior1, 2021-07-24:
@robertvorthman what hardware are you using? (out of curiosity)

robertvorthman, 2021-07-24:
@WeekendWarrior1 Wemos D1 mini

WeekendWarrior1, 2021-07-24:
> @WeekendWarrior1 Wemos D1 mini

Sorry, what lights?

robertvorthman, 2021-07-24:
> > @WeekendWarrior1 Wemos D1 mini
> 
> Sorry, what lights?

WS2811 light strip. This one: https://www.amazon.com/dp/B07PJZR3MZ/

Joshfindit, 2022-04-26:
I support this for all lights: PWM, LEDC, Addressable, etc.
Especially as it relates to HomeAssistant: `0%` = off, `1%` = `defined_minimum`, `100%` = `defined_maximum`

WeekendWarrior1, 2022-04-26:
> I support this for all lights: PWM, LEDC, Addressable, etc. Especially as it relates to HomeAssistant: `0%` = off, `1%` = `defined_minimum`, `100%` = `defined_maximum`

https://esphome.io/components/output/index.html?highlight=zero_means_zero
The functionality works within ESPHome, requires the `zero_means_zero` flag set to `true` on the output component

Joshfindit, 2022-04-26:
Yes, that works for the Output component, but not any other type of light.

WeekendWarrior1, 2022-04-27:
> Yes, that works for the Output component, but not any other type of light.

Apologies, you are totally correct.

burnbabyburn, 2022-09-15:
i hereby support this feature-req.
Since this issue celebrates his second year of existance there will probably never be a fix :(

mbrevda, 2022-11-13:
Seeing a seemingly similar issue with Athom bulbs: < 10% in HA turns off the led.

Janldeboer, 2022-12-08:
I'm currently working on this feature.
What I found out during this, is that (at least when using WS2812b), the gamma correction is responsible for the flickering.
Setting the parameter `gamma_correct` to 1 (defaults to 2.8 when using addressable LED-strips) leads to clean colors on every level of brightness.

xuefer, 2022-12-09:
Should the effect of gamma correction be compress to 1 when brightness is
closing to low?

Janldeboer, 2022-12-09:
> Should the effect of gamma correction be compress to 1 when brightness is closing to low?

I did try that at some point, but then colors like an almost pure red with just a little bit blue will turn into purple when to turn down the brightness.

Janldeboer, 2023-01-06:
A bigger issue that I see here is that we have an `ESPColorCorrection` class which is only used by `AddressableLight`, while other light components accomplish color correction by using the `current_color_as_...` methods provided by the `LightState`.

To implement min/max brightness, we currently would have to implement that into both.
That doesn't sound like a good practice to me, so I will try to combine the too concepts into one.
In my opinion, the  `ESPColorCorrection` should be used everywhere as is works with pre-calculated tables that save a lot of computational.

1Reed, 2023-01-29:
Any updates here? I use a WS2812b LED-Matrix for a coustom light and have the Problem that the light only turns on, if the brighness is settt to 11% (or higher). Hope it will be implemented and an UP for working on it @Janldeboer!