esphome/feature-requests #921: make global variables editable via webserver component

pille, 2020-09-30:
**Describe the problem you have/What new integration you would like**

using global variables is a good way to correct/complete data on the device and parameterize sensors.
sometimes those parameters need to be changed, which is currently not possible on-the-fly (without re-flashing or extra hardware/interface).

***minimal***
please add a generic way to expose global variables to the webserver component and make them _editable_.
an additional option may be introduced to mark a variable as editable on its definition. when this option is set and the webserver component is activated an appropriate _input field_ is shown on the web interface, that allows setting a new value.

***extended***
this would be a minimal implementation of course more comfort is possible, e.g. providing helper buttons like "plus", "minus" or "increase", "decrease" for a better user interface for known variable types like ```int``` or a _checkbox_ for ```bool```.

variable config options themselves could be altered, too. e.g. ```restore_value```, ```initial_value```.

**Please describe your use case for this integration and alternatives you've tried:**

my gas meter measures throughput and needs an offset added to make it congruent with the official counter.
missed update cycles make the meter deviate more and this feature would make a correction more convenient, than re-flashing the whole device, while after fixing ```initial_value```.

**Additional context**

skylord123, 2021-07-19:
Yeah this would be extremely helpful.

I'm building a timer for turning on/off grow lights and it must work even if the network is down. My ESP device has it's own RTC module for storing and keeping time (which syncs with an internet source when available). I want to be able to store what time the lights turn on and turn off in a global variable and make it so I can set this via the web interface as well as from an API request.

mmccoo, 2022-05-02:
With a more recent version, this appears to be possible.

I'm not an esphome developer, but I stumbled on this ticket yesterday. The code below needs updates to the webserver that happened in the last year. Before I updated my esphome install, I didn't see the slider for the number.

You can define a value:
number:
  - platform: template
    name: "Template number"
    optimistic: true
    restore_value: true
    min_value: 0
    max_value: 100
    step: 0.5
    id: the_number

and then reference it in your automations:
on_press:
      - switch.turn_on: relay
      - delay: !lambda "return id(the_number).state*1000;"
      - switch.turn_off: relay

I'm making an exposure timer for silkscreen printing.

closed 2022-07-06 by nagyrobi

skylord123, 2022-07-06:
That is nice we have something for numbers. Would be nice to have something for setting a variable to a string or boolean.

nagyrobi, 2022-07-06:
Use a template switch for boolean...

skylord123, 2022-07-06:
> Use a template switch for boolean...

I realized that after I posted it which is why I modified my comment.

Still, string support would be really nice. Especially for devices utilizing a display.

nagyrobi, 2022-07-07:
Possible with [Home Assistant Text Sensor](https://esphome.io/components/text_sensor/homeassistant.html) pointing to an [Input Text](https://www.home-assistant.io/integrations/input_text/).

skylord123, 2022-07-07:
> Possible with [Home Assistant Text Sensor](https://esphome.io/components/text_sensor/homeassistant.html) pointing to an [Input Text](https://www.home-assistant.io/integrations/input_text/).

Hmm, that may actually work pretty well. I'm assuming I will have to map it to a global variable in order to have it persist and be available when the device restarts/network is unavailable.

nagyrobi, 2022-07-08:
Not even necessarily, HA will keep the `input_text` values for the node, which when connected will be resynced from HA when node restarts.