esphome/feature-requests #924: Display Component - Get current page

JensGuckenbiehl, 2020-09-30:
**Describe the problem you have/What new integration you would like**

I would like to use the id or index of the currently displayed page in an expression

**Please describe your use case for this integration and alternatives you've tried:**

I find the possibility to show different pages on one display very good. But I would like to have 2 buttons to perform different functions depending on the currently displayed page. But I there is no way to return the current page to use it in a expression for the buttons.

**Additional context**

nagyrobi, 2022-07-08:
Implemented with `display.is_displaying_page` condition:
https://esphome.io/components/display/index.html#display-pages

closed 2022-07-08 by nagyrobi

escoand, 2022-07-08:
With lambda you could do this as well:
```yaml
switch:
  - platform: template
    name: Uhr
    icon: 'mdi:clock-digital'
    lambda: 'return id(ledmatrix).get_enabled() && id(ledmatrix).get_active_page() == id(page_clock);'
    turn_on_action:
      - lambda: 'id(ledmatrix).set_enabled(true);'
      - display.page.show: page_clock
      - light.turn_on: ledstrip
```

bachya, 2022-09-20:
FYI, I achieved this idea by building a map of display page pointers to strings:

```yaml
page_id: !lambda |-
  std::map<const esphome::display::DisplayPage*, std::string> map {
    { id(display_page_page1), "page1" },
    { id(display_page_page2), "page2" },
    { id(display_page_page3), "page3" },
  };
  return map[id(main_display).get_active_page()];
```