esphome/feature-requests #925: Read NDEF in NFC tags with PN532

timotoots, 2020-10-01:
**Describe the problem you have/What new integration you would like**
NFC tags contain NDEF fields that would be interesting to use.

**Please describe your use case for this integration and alternatives you've tried:**
You can URL and trigger something by the URL. For example play a sound file on a media player.

**Additional context**

I have posted feature request also to Home Assistant.
https://community.home-assistant.io/t/add-support-for-ndef-fields-in-tags-sections/228536

Active NDEF library for Arduino:
https://github.com/Flos/NDEF

timotoots, 2020-10-03:
It seems there is a pull request to another project that could be used.

jesserockz, 2020-10-04:
I am working on this locally (for the tagreader project actually) but I have not pushed my code anywhere yet. (I might push it up as a draft PR soon)

jesserockz, 2020-10-04:
esphome/esphome#1316

closed 2021-01-14 by jesserockz