esphome/feature-requests #926: PN532 tag reporting

timotoots, 2020-10-01:
**Describe the problem you have/What new integration you would like**
 It would be great to configure if the reader publishes each scan or only tag_on and tag_off events.

**Please describe your use case for this integration and alternatives you've tried:**

There are cases when the tag is placed on the reader and kept there.

**Additional context**

napieraj, 2020-10-03:
what is the difference between this and #925?

napieraj, 2020-10-03:
the [pn532 binary sensor](https://next.esphome.io/components/binary_sensor/pn532.html#nfc-rfid-tag) offers tag-on/tag-off tracking, use the `on_press` and `on_release` automations of that component -- they will have the same behavior as your proposed tag_on and tag_off events

timotoots, 2020-10-03:
Yes, it seems perfect! But if I understand correctly the UID for specific tag is required. It would usable if it worked for any tag as well.

napieraj, 2020-10-03:
I will reopen the issue, feel free to modify the title and description of your feature request to specifically the functionality you would like to see, the most logical is an extension to the binary_sensor that triggers on all tag UIDs

DefenestratingWizard, 2025-03-19:
Sounds like you want an in_field sensor? Is this correct?