esphome/feature-requests #928: Fan triggers on_turn_on, on_turn_off

epiniguin, 2020-10-02:
**Describe the problem you have/What new integration you would like**
It would be good to have on_turn_on/on_turn_off triggers for fan.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

napieraj, 2020-10-03:
which component are you referring to? binary fan? speed fan?

evasil, 2020-10-03:
I would like to ask for the same feature.

Use case: 

My ESP2866 (Sonoff SV) is controlling the fan by PWM and also provides the 12v power to the same fan via relay.
Instead of having relay always to be ON (with PWM at 0% when fan is not needed to run) it would be better to have the trigger when fan is on (or pwm changed) -> check if relay is on and if not then turn on the relay.

epiniguin, 2020-10-03:
>  which component are you referring to? binary fan? speed fan?

Fan core.
It should be like light triggers.

napieraj, 2020-10-03:
a workaround is using a template output as a sort of man-in-the-middle to get the current state and act upon it, before passing on the output value to the destination output (be it binary or float) -- here's an example for a binary fan attached to a GPIO output with a template output in the middle to offer you the `on_turn_on/off` functionality

```yaml
fan:
  - platform: binary
    output: fan_template_output
    name: "My Fan"

output:
  - platform: template
      id: fan_template_output
      type: binary
      write_action:
        - if:
            condition:
              lambda: return state;
            then:
              - output.turn_on: fan_gpio_output
              - logger.log: "fan was switched on"
            else:
              - output.turn_off: fan_gpio_output
              - logger.log: "fan was switched off"
  - platform: gpio
    pin: 32
    id: fan_gpio_output
```
I haven't tested this but it should work

epiniguin, 2020-10-04:
> a workaround is using a template output

I am using this workaround. It's clear.
But it would be better with triggers.
Before 0.115 I used templates with lights also. But now for lights there are triggers, and I retrofitted my yamls.

napieraj, 2020-10-04:
I would instead propose something like a template fan, with both binary (on/off) and multiple speed modes, and actions for all the aforementioned

epiniguin, 2020-10-04:
I use this template. It works without problem.
But with triggers it would be better.

`- platform: template
    name: "Fan1 dummy switch"
    internal: true
    lambda: |-
      return (id(fan_1).state);
    on_turn_on:
      then:
        ......
    on_turn_off:
      then:
         ......
`

loongyh, 2021-04-04:
Maybe also consider adding `.is_on`/`.is_off` conditions for fan?

jesserockz, 2021-08-31:
Implemented in esphome/esphome#1726

closed 2021-08-31 by jesserockz