esphome/feature-requests #930: Cover, open_duration and close_duration, variables instead of just numbers

osalval, 2020-10-03:
**Describe the problem you have/What new integration you would like**
I want to be able to vary from Home Assistant, the number of seconds of:
"open_duration" and "close_duration" variables for a Cover type: "Time-Based Cover", using for example an "input_number"

**Please describe your use case for this integration and alternatives you've tried:**

 I have tried this only for **open_duration**, using  id: **${name}_open_duration**, like this:

`sensor:
  - platform: homeassistant
    name: "Tiempo de Subida"
    id: ${name}_open_duration
    entity_id: input_number.persiana_habita_set_uptime
`

`cover:
  - platform: time_based
    device_class: shutter
    name: "Persiana Habitación"
    id: ${name}_cover
    open_action:
      - switch.turn_on: open_cover

    open_duration: 
      seconds: ${name}_open_duration

    close_action:
      - switch.turn_on: close_cover
    close_duration: 23s

    stop_action:
      - switch.turn_off: open_cover
      - switch.turn_off: close_cover
` 

**Additional context**
I can't Validate, I get the error: Expected float

napieraj, 2020-10-03:
those values are set on compile-time, before there is even a connection to Home Assistant -- you can update them manually using a lambda and the `.set_open_duration()` and `.set_close_duration()` arguments of the time-based cover component: https://esphome.io/api/classesphome_1_1time__based_1_1_time_based_cover.html

ludrao, 2021-12-12:
Hi @napieraj , not sure how it is generated but https://esphome.io/components/cover/time_based.html does not seem to document these two lambdas ?

nagyrobi, 2022-07-06:
You can use lambdas as @napieraj suggests.

closed 2022-07-06 by nagyrobi