esphome/feature-requests #932: Support for MS8607 PHT (Pressure Humidity Temperature) sensor

e28eta, 2020-10-04:
**Describe the problem you have/What new integration you would like**

I'd like built-in support for the I2C-based MS8607 Pressure, Humidity, Temperature sensor. https://www.adafruit.com/product/4716

**Please describe your use case for this integration and alternatives you've tried:**

I believe this is supportable via a custom I2C component, but I think this probably deserves to be built-in?

**Additional context**

I'm new to esphome (awesome project :tada:), and intend to poke around at supporting this. I'm not sure what's involved yet, but I'm hoping this is supportable.

e28eta, 2020-10-12:
I believe I'm making good progress on this.

Something about the MS8607, which I haven't found another example of, is that it has **two** I2C addresses. One for Temperature/Pressure, and the other for Humidity.

My current approach looks something like this:

```yaml
MS8607Component = ms8607_ns.class_('MS8607Component', cg.PollingComponent, i2c.I2CDevice)

CONFIG_SCHEMA = cv.Schema({
    cv.GenerateID(): cv.declare_id(MS8607Component),
    # TODO: can/should these be optional to ignore sensors we don't care about?
    cv.Required(CONF_TEMPERATURE): sensor.sensor_schema(UNIT_CELSIUS, ICON_THERMOMETER, 1),
    cv.Required(CONF_PRESSURE): sensor.sensor_schema(UNIT_HECTOPASCAL, ICON_GAUGE, 1),
    cv.Required(CONF_HUMIDITY): sensor.sensor_schema(UNIT_PERCENT, ICON_WATER_PERCENT, 2)
        # Humidity is at a different I2C address, but same physical device/I2C bus
        .extend(cv.Schema({ cv.Optional(CONF_ADDRESS, default=0x40): cv.i2c_address }))
}).extend(cv.polling_component_schema('60s')).extend(i2c.i2c_device_schema(0x76))
```

In the c++ class, I create a 2nd `I2CDevice` for humidity as a `protected` member of `MS8607Component`, using the component's `parent_` and the humidity address from the config. I don't think there's any reason to allow full configuration of the `i2c.i2c_device_schema`, since there's only one set of SDA/SCL pins.

IDK if feature requests commonly receive pre-implementation feedback, but it's welcome! I'm also happy to wait until PR review, and make any changes then. So far, this is the primary surprise, and everything else seems pretty straightforward.

e28eta, 2020-12-22:
WIP branch is https://github.com/e28eta/esphome/tree/ms8607

e28eta, 2021-06-21:
I’ve had some flakiness on cold boot where the sensor is unavailable, but haven’t been motivated to figure out why, since it mostly works for me and a reset is sufficient to fix 🙁 

maybe a race, or a bad timeout, or a bad init? Like I said, I’m not sure.

freemanr7, 2021-07-10:
Thanks for sharing your work and discovery.  I have the same desire to add this sensor.  I have used the MS5611, changed the I2C address, and I have temp data and pressure data coming in.  The pressure data is wrong, as the calculations are different between the two sensors.    I'm pretty new at this.  Wondering if there is some place I can go to edit the calculations, or create new.

e28eta, 2022-03-26:
Hey @freemanr7,

I don't know if you saw :point_up: the PRs I've created. I also don't know how quickly to expect them to be reviewed, but one thing that might be useful is if you had a chance to check it out. I realize you may have moved on since it's been almost 9 months, but... 🤷‍♂️ It sounds like you had the hardware connected to your i2c bus correctly. On the config side, this was sufficient for me:

```yaml
# Tell esphome to use currently un-merged ms8607 PR
external_components:
  - source: github://pr#3307
    components: [ ms8607 ]

# You may have other sensors, if so just add ms8607 platform
sensor:
  - platform: ms8607
    temperature:
      name: Temperature
    humidity:
      name: Humidity
    pressure:
      name: Pressure
```

You can preview the [ms8607 docs](https://deploy-preview-1974--esphome.netlify.app/components/sensor/ms8607.html), and obviously feel free to provide different names for the temperature/humidity/pressure sensors - they should work like any other esphome sensor.

As far as I can tell from the datasheet, it's not possible to have different i2c addresses for the chip than the defaults, but if they're not `0x76` + `0x40` you'll need to fix that too.

To my knowledge & limited testing, it's working great now. I've updated my esphome install to the public v2022.3.0 version, and have a working device using [`external_components`](https://esphome.io/components/external_components.html) and esphome/esphome#3307

closed 2024-03-04 by jesserockz