esphome/feature-requests #933: Waveshare - Partial update support for 4.2

Wolbyworld, 2020-10-04:
**Describe the problem you have/What new integration you would like**

The 4.2inches waveshare screen don´t currently support the partial update feature (implemented using full_update_every for the 1.54in, 2.13in) however the hardware allow to do it. An Arduino user using a ESP32 is able to do it, and shows how in [this video](https://www.youtube.com/watch?v=mySnfyyL4i4)

**Please describe your use case for this integration and alternatives you've tried:**

I am currently using the full update with 1min change. The problem is that it looks terrible and don´t allow higher refresh frequency. 

I could code the .ino sketch and upload it, but then the integration with the other sensor would be significantly more painfull.

**Additional context**

anekinloewe, 2021-01-03:
I thing this guy explain really good how partial update for the 4.2" display works: 
https://benkrasnow.blogspot.com/2017/10/fast-partial-refresh-on-42-e-paper.html
He share example code, too. Unfortunately my c++ is too bad to implement it myself.

JonathanTreffler, 2021-02-27:
This would be great feature.
But it is a duplicate of #250 and that didn't lead anywhere.

DerWe, 2021-05-16:
I recently build a Display with a 4.2 in Waveshare Panel as well. Standing on a bookshelve, partial Update would be a nice feature to let the refresh be less annoying.

Warter21, 2021-06-18:
yes, it would be a pretty nice feature.

mbo18, 2021-08-25:
I tried to understand how it works to add this to ESPHome but failed as I’m not a dev… still no one to work on this?

Buuuh, 2022-02-13:
Also still waiting for this. 

At least i found a custom solution for the 2.7" display. A bigger one would be nice, but only full update is a no go (and also not required) for my usecase, because the flickering is anoying. 

I don't get why it is not implemented... 

No partial update, no buy.

atoomkern, 2022-03-22:
> Also still waiting for this.
> 
> At least i found a custom solution for the 2.7" display. A bigger one would be nice, but only full update is a no go (and also not required) for my usecase, because the flickering is anoying.
> 
> I don't get why it is not implemented...
> 
> No partial update, no buy.

Can you share your custom solution for the 2.7" display? I have the 2.7" display but can't get it to partial update

Buuuh, 2022-03-22:
> Can you share your custom solution for the 2.7" display? I have the 2.7" display but can't get it to partial update
https://github.com/elad661/rpi_epd2in7

malkie-hass, 2024-01-07:
I would love to see this feature. WIth the full refresh this is taking to long. I have the waveshare 4.2 to show some temperatures from my meater bbq thermometer. And somehow this was not what I expected how it works :(