esphome/feature-requests #936: Reading text textboxe and global variables from Nextion displays.

nrodgers1987, 2020-10-06:
**Describe the problem you have/What new integration you would like**

I would like to be able to get global variable from my nextion display into ESPHome
**Please describe your use case for this integration and alternatives you've tried:**

Im creating a temperature station to use at work and free up some of our staff from having to take temperatures of our employees manually. The nextion display has a number pad that builds a string so I can get the employee number of the person getting their temp taken, then it asks two questions, The esp will use an infrared temperature sensor to get the person's temperature. Then it will post the info from the nextion display and temperature to our internal api which writes into our sql server. I have everything figured out except retrieving the variables nextion is creating. I can see many other uses for this ability for myself and others in the future. I'd also like to post the build I do when I'm done so other businesses or anyone really can use this instead of keeping track of everything on a piece of paper
**Additional context**

Woutch, 2020-10-26:
Hi, also following this. 

I created a keypad on my nextion display for entering my security code and want to arm/disarm my alarm, but I need a way to get this value back to esphome. 

Now the keypad enters the numbers into a txt field, but I need to be able to read it if I press disarm.

Many thanks!!

nrodgers1987, 2020-10-27:
@Woutch I had to set up 1 binary sensor per button i wanted to get into ESPHome. This will be much easier when variables and textboxes can be read from ESPHome. Make sure you check "[send component id](https://uptimefab.com/wp-content/uploads/2020/02/send-component-ID.png)" on press or release in the Nextion software. Below is how i went about it. I hope this helps you for now.

```
#####-Global Variables-#####
globals:
  - id: emp_id
    type: std::string
    restore_value: no
    initial_value: ''
  - id: keypad_int_count
    type: int
    restore_value: no
    initial_value: '0'
  - id: keypad_int_count_max
    type: int
    restore_value: no
    initial_value: '4'

#####-Nextion Page1 Keypad Screen-#####
  - platform: nextion
    page_id: 1
    component_id: 1
    name: "Nextion keypad 1"
    id: keypad_1
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "1";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 2
    name: "Nextion keypad 2"
    id: keypad_2
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "2";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 3
    name: "Nextion keypad 3"
    id: keypad_3
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "3";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 4
    name: "Nextion keypad 4"
    id: keypad_4
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "4";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 5
    name: "Nextion keypad 5"
    id: keypad_5
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "5";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 6
    name: "Nextion keypad 6"
    id: keypad_6
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "6";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 7
    name: "Nextion keypad 7"
    id: keypad_7
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "7";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 8
    name: "Nextion keypad 8"
    id: keypad_8
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "8";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 9
    name: "Nextion keypad 9"
    id: keypad_9
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "9";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 10
    name: "Nextion keypad cancel"
    id: keypad_cancel
    on_release:
     then:
  - platform: nextion
    page_id: 1
    component_id: 11
    name: "Nextion keypad 0"
    id: keypad_0
    on_release:
     then:
     - lambda: |-
          if (id(keypad_int_count) < id(keypad_int_count_max)) {
            id(emp_id) = id(emp_id) + "0";
            id(keypad_int_count) = id(keypad_int_count) + 1;
          }
  - platform: nextion
    page_id: 1
    component_id: 12
    name: "Nextion keypad next"
    id: keypad_next
    on_release:
     then:
```

SenexCrenshaw, 2021-01-12:
Have a look at the following PR: esphome/esphome#1464
This PR adds a lot of functionality!

nagyrobi, 2022-07-06:
Implemented wit PR above.

closed 2022-07-06 by nagyrobi