esphome/feature-requests #939: Component to make YAML source code available on devices

Johnny-Larue, 2020-10-07:
**Describe the problem you have/What new integration you would like**

I'd like a mechanism to recover the actual YAML source for the device from the device itself.

**Please describe your use case for this integration and alternatives you've tried:**

In the event of the loss of the original YAML files, it would be nice to be able to recover the original YAML from the device itself, perhaps in an accordion box on the Web Server page, or in a dialogue.

At the very least, it would be great to be able to download a compressed version of the text, at the very best, simply be presented with the raw code in a PRE HTML tag, with a 'Copy' button to copy the code entire as simple text.

If possible a version of the code with the substitutions from the secrets.yaml (and any other source) clearly visible would be ideal as well. Perhaps this could be an option in the eventual component section in the YAML source.

**Additional context**

Example of what I'd love to be able to do (I've attached two mock-ups for reference):

```
web_server:
  port: 80
  yaml_include: True
    reveal_secrets: True
    reveal_substitutions: True
```

All of this will, of course, increase the size of the resulting .bin file, but I for one wouldn't mind the increased burden on my devices in order to have this available to me. Additionally, this does present a rather large security risk, but the fact that it's optional should mitigate that somewhat. Perhaps requiring credentials for the Web Server would be prudent?

[image]
[image]

dentra, 2021-05-28:
You could try my solution:
https://github.com/dentra/esphome-components/tree/master/components/backup

nagyrobi, 2022-07-07:
Duplicate of https://github.com/esphome/feature-requests/issues/1507

closed 2022-07-07 by nagyrobi